pymeasure/pymeasure #19: 1 process or thread for each instrument

guenp, 2016-02-21:
Hi @cjermain,

I've just started using PyMeasure at the PPMS set-up (Windows PC).

One thing I've been doing in my measurement code, is for each instrument, run a separate server process (```InstrumentDaemon```) that is dedicated to talking to that instrument. I then create a pipe for each ```InstrumentDaemon```, and talk to the pipe using a ```RemoteInstrument```. That way, you circumvent the pickling issue (having to reload the instrument driver in the ```Procedure``` class), and you also have the possibility to get datapoints asynchronously.

I've tested my code with PyMeasure and works fine - not perfect yet, I have some ideas to improve it :sweat_smile: But see the attached screenshots to get an idea of the workflow. I now pass a list of pipes to the ```Procedure``` object, which then uses it in the measurement, but this could be done more smoothly (with a global variable or something rather).

What do you think? Shall I start a branch/PR and start implementing this stuff into PyMeasure?

[image]
[image]
[image]

cjermain, 2016-02-22:
Thats a great idea -- a PR is welcome. Can we implement this without affecting the standard instrument usage? Are you thinking of using threads or processes?

guenp, 2016-02-22:
Great! Yes, it will definitely be backwards compatible. Do you have any preference between threads vs. processes? Do pipes work with threads as well?

cjermain, 2016-02-23:
Since threads are localized to specific processes, going that route requires identifying a "master process" for the threads to be spawned from. Would this be the `Worker` in that case or something else? If they are each an independent process, it avoids this hierarchy. So I would go with processes, unless you see any other complications that threads simplify. Threads should work just as well with pipes.

guenp, 2016-02-23:
Hmm, I'm now thinking 1 ```Instruments``` process for all instruments, then 1 thread per instrument...
I'll try out some things!
Also, yay for pymeasure going public :)

guenp, 2016-04-12:
Hi @cjermain,
I'm now looking into using ```ZMQ``` for instrument multi threading. Apparently sockets are much faster than ```Queue```s. :smiley: Basically, I'm extending the ```Instrument``` class to inherit stuff from ```Listener``` and run a loop that listens to the socket for commands: [instrumentdaemon.py](https://github.com/ralph-group/pymeasure/blob/b8109f2fd47ca89ca28dd27f00400db2c78532c4/pymeasure/instruments/instrumentdaemon.py) (still work in progress). This way, instrument parameters can be acquired asynchronously. What do you think? Also, I was wondering - the ```ZMQ``` stuff you've implemented so far (for ```Worker```) is not specifically being used atm, right? Would it be an idea to move from ```Queue```s to ```ZMQ``` sockets (e.g. instead of ```monitor_queue``` and ```recorder_queue``` listen to the ```ZMQ``` socket) or is there a specific reason you haven't done that?

cjermain, 2016-04-12:
The big pitfall of using ZMQ ([publisher/subscriber](http://zguide.zeromq.org/py:all#Getting-the-Message-Out)) is that both parties in the communication may not be ready, so messages can be lost. For example the Recorder may only start listening for messages after the Worker has published a few data points, which will not be received. Queues store the data until a pop call, so you will never miss packets or get out of sync. That is primarily why I kept with Queues. I also wanted to minimize the required packages for PyMeasure, so I made ZMQ optional.

That being said, ZMQ is very fast (especially with msgpack). We could replace Queues entirely if a good handshaking algorithm is implemented, so we can confirm that both sides are connected. 

This is where having good unit tests that confirm our existing behavior is reproduced is important. I've set up automatic testing with `py.test` using TravisCI and AppVeyor to evaluate Linux (TODO: add MacOSX) and Windows. As we add major features like this, it would be great to have good tests to confirm things are working. During the pull request you will see the results of these tests. I recommend starting a pull request the moment a new branch is started, so that discussion can be directed to the code as it evolves and tests can be viewed.

guenp, 2016-04-12:
Ok - thanks for the feedback! I've started a PR & I'll try to implement the handshake.
FYI - I'm now thinking it could either be an idea to have a separate process for each instrument, or a process per interface (GPIB, TCP, USB) and have threads for each instrument within those processes (because e.g. acquiring through GPIB is sequential and having multiple cores won't make a difference anyway and will just eat up memory). I'll do some tests and see which case would be optimal in terms of acquisition time. Thoughts/suggestions are of course welcome.

cjermain, 2016-04-13:
I agree that you should limit the number of processes, and your suggestion to have one per adapter makes sense. Perhaps the instruments can be threads and the adapters can be processes. I'll have to think about that...

asteppke, 2017-03-07:
Just for reference, a similar project, [Pythics](https://github.com/dursobr/Pythics), has had great success with using separate processes instead of threads. This also allows the coupling to be much weaker, imagine a python process for one instrument than can either talk to a Pymeasure master process or alternatively to some other software. Additionally we have now several instruments in our lab that contain basically an embedded PC. Here it would be nice to have a process running on the embedded PC for instrument control, while the managing and data processing happens on a different PC. 
This would be harder or impossible if all the instrument drivers are running as threads in a single process.

cjermain, 2017-03-08:
Interesting point @asteppke. Using a process running on the embedded PC is how we are planning to implement the Quantum Design PPMS (PR #49). So far we don't have a general Adapter for this case. Would you use one if it was available?

bilderbuchi, 2022-03-20:
This is quite the old issue, and I think a number of things have changed in the background regarding the parallelism setup since then. I'll close this, please add another issue if you should have a similar need.

closed 2022-03-20 by bilderbuchi