pymeasure/pymeasure #28: Plotting functions for Worker class

minhhaiphys, 2016-06-02:
Not sure how everyone uses the Experiment class, but it seems to me that the plotting function can be incorporated into the Worker so that the Worker can take care of executing Procedure, handling Result and plotting at the same time. What do you think?

Another issue related with the plotting functions currently in the Experiment class is that they were written mainly for jupyter notebook. I have rewritten part of it (not yet put into PyMeasure) to use matplotlib to run on console. If everyone is interested in reviving the plotting functionality, please brainstorm here. (We can use pyqt as well, which can be faster).

cjermain, 2016-06-02:
The purpose of the Worker class is to execute the procedure -- thats it. The reason for this separation of concerns is that the Procedure is a synchronous sequence of events. If the Worker also handles the Results and plotting, the Procedure will have to block to wait for these operations to execute. The plot and any other graphical tools are asynchronous, which need to react to user input. We've put in a lot of work to get PyMeasure to its current state, where it handles both the asynchronous and synchronous sections properly, making graphical interfaces easy to setup. Does that make sense? I recommend looking through the `experiment.Manager` class and the examples to understand the reasons for this design.

I don't recommend using `matplotlib` for any plotting that is not inside the Jupyter notebooks. We have found the performance to be quite poor compared to pyqtgraph, which is the plotting library we use in PyMeasure. I've already made some convenient classes for dealing with graphing -- see `display.curves.ResultsCurve` and `display.curves.BufferCurve`. Please go through the `display` module before writing any new code.

cjermain, 2016-06-02:
To be clear, the `experiment.Experiment` class should only be used in Jupypter Notebooks. Check out the [`gui.py`](https://github.com/ralph-group/pymeasure/blob/master/examples/gui.py) example that does not use notebooks.

minhhaiphys, 2016-06-04:
Thanks for the explanation, Colin! And sorry for not checking through all the relevant docs. I found the comment: 

> Live plotting loop for jupyter notebook, which automatically updates
        (an) in-line matplotlib graph(s)

in the `def plot_live` of `experiment.Experiment` class, so a little confused if the class was intended for use on jupyter notebook only. Might be better to move that comment somewhere near the top of the file?

closed 2016-06-04 by minhhaiphys