pymeasure/pymeasure #29: Instruments as Procedure's local attributes

minhhaiphys, 2016-06-02:
It happened to me that the Procedure class could not execute when it contained Instrument instances as its attributes. Sorry, the probe station is being used by others so cannot re-run and get the error messages, but it seemed the problem lied in the __new__() function of the VISAAdapter class when it tries to choose parent class depending on the Visa version.

Temporarily I use Instrument instances as global and the issue disappeared.

cjermain, 2016-06-02:
Can you give a code example that produces the issue?

minhhaiphys, 2016-06-04:
I've narrowed down the issue. If I use `instrument.Instrument` instances as global, or as class properties as shown in the file "testProcedure_1.txt" shown below, it was fine.

[testProcedure_1.txt](https://github.com/ralph-group/pymeasure/files/298919/testProcedure_1.txt)

The error occurs if I tried to put them as a `procedure.Procedure' instance's attributes as shown in the file "testProcedure_2.txt" shown below.

[testProcedure_2.txt](https://github.com/ralph-group/pymeasure/files/298920/testProcedure_2.txt)

And below are the error messages:

> INFO:root:Set up Results
12:00:22 PM: Set up Results (root, INFO)
INFO:root:Created worker for TestProcedure
12:00:22 PM: Created worker for TestProcedure (root, INFO)
INFO:root:Starting worker...
12:00:22 PM: Starting worker... (root, INFO)
Traceback (most recent call last):

  File "<ipython-input-1-c9f18df5cb0a>", line 1, in <module>
    runfile('C:/Users/BuhrmanD20GMW/Documents/instrumentation/testProcedure_2.py', wdir='C:/Users/BuhrmanD20GMW/Documents/instrumentation')

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 807, in runfile
    execfile(filename, namespace)

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 95, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)

  File "C:/Users/BuhrmanD20GMW/Documents/instrumentation/testProcedure_2.py", line 88, in <module>
    worker.start()

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\multiprocessing\process.py", line 105, in start
    self._popen = self._Popen(self)

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\multiprocessing\context.py", line 212, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\multiprocessing\context.py", line 313, in _Popen
    return Popen(process_obj)

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\multiprocessing\popen_spawn_win32.py", line 66, in __init__
    reduction.dump(process_obj, to_child)

  File "C:\Users\BuhrmanD20GMW\Documents\WinPython-64bit-3.4.3.7\python-3.4.3.amd64\lib\multiprocessing\reduction.py", line 59, in dump
    ForkingPickler(file, protocol).dump(obj)

PicklingError: Can't pickle <class 'pymeasure.instruments.instrument.Instrument.__init__.<locals>.Object'>: attribute lookup Object on pymeasure.instruments.instrument failed

cjermain, 2016-06-06:
This issue comes up as a result of using processes. When the Worker is started, it spawns a new process which does not have any variables initialized. Python Pickles everything and sends them to the new process, which can cause issues like we saw in #12. Typically to initialize instruments, I just put them in the startup method. You can also call out to a different function, or use Parameters to pass in the GPIB addresses. 

```python
class TestProcedure(Procedure):

    DATA_COLUMNS = ['Step', 'Output']

    def startup(self):
        log.info("Starting up...")
        self.output_instr = Keithley2000(26)
        self.input_instr = Yokogawa7651(2)

```
This is an error worth more investigation, but I suggest using a workaround for now.

bilderbuchi, 2020-10-26:
@cjermain @minhhaiphys this is quite an old bug, I think there were some changes to the process/threading architecture - does that still occur?

bilderbuchi, 2022-01-09:
There has been no update on this bug in a long time, so I'll close this. feel free to reopen if you disagree.
If you encounter such a problem, please open a fresh bug report!

closed 2022-01-09 by bilderbuchi