pymeasure/pymeasure #38: Creating conda-forge packages for pymeasure

melund, 2016-08-03:
Pymeasure looks awesome. I can't wait to get a chance to use it for something. 

Until then I thought I could contribute by adding a conda-forge package for it. https://conda-forge.github.io/

The first step was to make a `pyvisa`1.8 conda forge package.  That went live yesterday. 
https://github.com/conda-forge/pyvisa-feedstock

So you can now install pyvisa with:
``` 
conda config --add channels pyvisa
conda install pyvisa
```

Next step is the `pymeasure` package. It is still under review (as a staged recipe) 
https://github.com/conda-forge/staged-recipes/pull/1175
It is still giving me some issues, but I will post them as separate issues. 

If you want to be added as maintainers for the `pyvisa` and `pymeasure` conda-forge packages just let me know.

cjermain, 2016-08-03:
Great idea @melund, thanks for bringing this up! PyMeasure only supports Python 3, and is specifically designed for 3.4 and 3.5, where we take advantage of the more recent multiprocessing library for concurrent execution. We fully switched to Python 3 in version 0.2 to remove the work of backporting, and are committed to [the Python 3 Statement](http://python3statement.github.io/).

Does conda-forge require 2.7 support?

melund, 2016-08-03:
@cjermain Yes. Sorry about that. I just got it mixed up with `pyvisa` which I also made a conda-forge package for. Python 2 is by no means required. We are also exclusive python 3 for [xonsh](http://xon.sh/)  (also on Python3 statements) 

I updated the recipe and it now builds fine. https://github.com/conda-forge/staged-recipes/pull/1175

melund, 2016-08-03:
Btw.  should I add you as co-maintainer of the package. You can also be added later, so no worries.

cjermain, 2016-08-03:
Nice work, thanks @melund! For a new release, do we just update `recipe/meta.yml` with the version number?

melund, 2016-08-03:
Yes. That is basically all there is to it. 
The  feedstock for the package is located here:
https://github.com/conda-forge/pymeasure-feedstock

I think the packages are still building at the moment. But when they are done you can just do:

```
conda config --add channels conda-forge
conda install pymeasure
```

ddale, 2016-08-04:
@melund would you please add me as a maintainer for pyvisa and pymeasure?

melund, 2016-08-05:
@ddale 

No problem. It is as simple as adding your GitHub name on the recipe. The conda forge system will then add you to the conda-forge organisation.

https://github.com/conda-forge/pyvisa-feedstock/pull/2
https://github.com/conda-forge/pymeasure-feedstock/pull/1

melund, 2016-08-05:
@ddale  I have merged the PRs now. So the system should soon add you. Let me know if it doesn't work

cjermain, 2016-08-09:
Looks like this is all set. Thanks @melund!

closed 2016-08-09 by cjermain

cjermain, 2016-08-22:
@melund or @ddale, can you add me as well to the conda-forge package? I'm planning to make a minor release (v0.4.2) tomorrow with updated instrument features and bug-fixes.

melund, 2016-08-22:
No problem. I made the PR.  @ddale can you merge this?

https://github.com/conda-forge/pymeasure-feedstock/pull/2