pymeasure/pymeasure #39: Syntax error on py 2.7

melund, 2016-08-03:
When creating the conda-forge packages I stubled on this in the build log: 

```
byte-compiling /opt/conda/envs/_build/lib/python2.7/site-packages/pymeasure/display/widgets.py to widgets.pyc
  File "/opt/conda/envs/_build/lib/python2.7/site-packages/pymeasure/display/widgets.py", line 229
    def __init__(self, *args, parent=None):
                                   ^
SyntaxError: invalid syntax
```

cjermain, 2016-08-03:
This is valid syntax in Python 3, but not 2.7. As mentioned in #38, only Python 3 is supported.

closed 2016-08-03 by cjermain

melund, 2016-08-03:
Ahh. Sorry about that. I thought I saw Python 2 support in `setup.py` but I think that probably `pyvisa`. No problem here. I use python 3 almost exclusively these days. 

I will update the conda-forge recipe to exclude py2