pymeasure/pymeasure #57: Feature: USBTMC Adapters and Instruments

SunPowered, 2017-02-28:
Some new instruments use the USBTMC protocol to communicate GPIB commands over a USB bus.  

An experiment I am conducting uses such an instrument and I have used the [python-usbtmc](https://github.com/python-ivi/python-usbtmc) library to extend the core `Adapter` and `Instrument` class.  

If there is interest, I will submit a PR when I feel its stable.

cjermain, 2017-02-28:
@SunPowered, that sounds great! Feel free to open a PR while working to discuss the feature.

TvBMcMaster, 2017-03-03:
#58 This is the code that I have been using to control my USBTMC instrument.  

Some things to note which may cause issues or confusion.

1.  In contrast to a serial port, the python USBTMC module must disconnect the device from the OS in order to communicate with it.  This means the device must be properly released back to the OS when python is finished with it.  When all things go well, this is not an issue.  During development, however, I was getting hang ups with the device that required a physical disconnect / reconnect of the USB cable.  This  happened when an unexpected error occurred in a thread, for example.  
2.  I have only tested this on a linux machine, I will try to run it in a Windows VM and see if I can get similar results.

bilderbuchi, 2021-03-03:
The associated PR was closed as stale some time ago, so I'll close this. I have a feeling that the PyVISAAdapter should be able to [deal with USBTMC connections](https://pyvisa.readthedocs.io/en/latest/introduction/names.html?highlight=usbtmc#visa-resource-syntax-and-examples), so before re-opening this, please try it out via PyVISA, i.e. pass an appropriate resource identifier to the Instrument.

closed 2021-03-03 by bilderbuchi