pymeasure/pymeasure #60: Appveyor Build Failing for Python34

TvBMcMaster, 2017-03-03:
The Appveyor [build](https://ci.appveyor.com/project/cjermain/pymeasure/history) is failing.

It started breaking at [build 0.3.170](https://ci.appveyor.com/project/cjermain/pymeasure/build/0.3.170) for build variables 
    
    PYTHONVERSION=3.4
    MINICONDA=C:\Miniconda3[-x64]

The latest build error is:

    UnsatisfiableError: The following specifications were found to be in conflict:
      - pyqt 4.11.4 py34_4 -> sip >=4.16.4,<4.18
      - sip 4.18 py34_0
    Use "conda info <package>" to see the dependencies for each package.

cjermain, 2017-03-03:
@TvBMcMaster, thanks for starting this issue. It looks like if we switch to the `pyqt-4.11.4-py34_7` package, `sip` 4.18 is supported. I'll start a PR to update the conda environment file.

cjermain, 2017-03-03:
This is fixed.

closed 2017-03-03 by cjermain