pymeasure/pymeasure #71: pip install pymeasure generates error

cbrown1, 2017-04-03:
[cbrown@cbrown2 ~]$ sudo pip2 install pymeasure
Collecting pymeasure
  Using cached PyMeasure-0.4.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-9TN0h5/pymeasure/setup.py", line 57, in <module>
        long_description=open('README.rst').read() + "\n\n" + open('CHANGES.txt').read(),
    IOError: [Errno 2] No such file or directory: 'CHANGES.txt'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-9TN0h5/pymeasure/
[cbrown@cbrown2 ~]$ sudo pip2 install pymeasure==0.4.2
Collecting pymeasure==0.4.2
  Using cached PyMeasure-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-62toxu/pymeasure/setup.py", line 55, in <module>
        long_description=open('README.rst').read() + "\n\n" + open('CHANGES.txt').read(),
    IOError: [Errno 2] No such file or directory: 'CHANGES.txt'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-62toxu/pymeasure/
[cbrown@cbrown2 ~]$

cjermain, 2017-04-04:
@cbrown1, thanks for reporting this. It looks like `CHANGES.txt` is not included in `MANIFEST.in`, so its not being rolled in with the package. I will fix this shortly.

I'm curious if this effects the wheel version. Do you have wheel installed?

cbrown1, 2017-04-04:
I can confirm that the wheel does install properly on python3, and also that the github version now installs on python2.

Thanks!

cjermain, 2017-04-04:
Great. PyMeasure only supports Python 3 to help [phase out Python 2.7](http://python3statement.github.io/). Are the pip issues specific to Python 2.7?

cjermain, 2017-04-05:
Thanks @cbrown1 for looking into this! Since this only affects the non-wheel pip installs, I will hold off on making a new release. I suspect that most users will have wheel installed, or be using the Anaconda distribution (since it includes the Qt libraries, which can be troublesome to install).

I will leave this open until the next release to inform anyone else who has this issue.

ghisvail, 2017-05-29:
The PyPI tarball is the canonical source for packaging by Linux distributions. Please reconsider the value of providing a sane PyPI tarball.

cjermain, 2017-05-29:
@ghisvail, this is already fixed on master and is just waiting on a new release. I wanted to wait until more substantial updates went through, but I'm open to making a new release if this is a show stopper for you. Keep in mind this bug only affects non-wheel pip installs. Given the Qt dependencies, I expect most users will go for the Anaconda distribution route and few will be using non-wheel pip. What do you think?

ghisvail, 2017-05-30:
> I'm open to making a new release if this is a show stopper for you.

It is currently a show-stopper for making a Linux package, but I took this opportunity to iron out some problems I discovered on the GitHub tagged release (#77, #78, #79, #80). Once these are all addressed, it would be nice to make a new release.

> Keep in mind this bug only affects non-wheel pip installs.

Quoting myself above: `The PyPI tarball is the canonical source for packaging by Linux distributions`. As a prospective packager, I am very much affected by this bug.

> I wanted to wait until more substantial updates went through

Tagging releases is quite cheap these days, so I fail to understand the value in waiting ([RERO](https://en.wikipedia.org/wiki/Release_early,_release_often)).

cjermain, 2017-06-04:
This is fixed in the newest release. Thanks everyone!

closed 2017-06-04 by cjermain