pymeasure/pymeasure #75: Writing results to multiple files

feinsteinben, 2017-05-27:
Hi,

In my set-ups, I'm writing the results to a primary location (network drive), and to a local backup location (local drive) - in case the network drive suddenly disconnects.

I'd like to know whether it's possible to do it with pymeasure. I tried to bind two Result objects to the same Procedure, but it doesn't seem to work.

Thanks,
Ben.

cjermain, 2017-05-29:
@feinsteinben, this is certainly possible. I would recommend implementing this by extending the [Recorder class](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/experiment/listeners.py#L83) to handle multiple data files. This can be passed correctly by extending the [Results class](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/experiment/results.py#L62) so that it can handle multiple `data_filenames`. That way you only need one Results class having multiple files to write to. What do you think?

feinsteinben, 2017-06-05:
sounds good. I'll try to do it during the weekend.

PYmeas, 2020-07-26:
Hello,

I can save multiples files generated from queued experiments from Keithley 2420 but i want to use a label _type_mode_ in the filename that is updated in the execute() from the Randomprocedure class. here i use fake data for testing purpose I am getting a NameError: name 'type_mode' is not defined. I understood that variable exit only within a class but i thought by inheriting Managedwindow from Randomprocedure this could be possible? 

``` ruby
import logging
log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())

import sys
import tempfile
from tempfile import NamedTemporaryFile
import random
import datetime
from datetime import datetime as dt, timedelta
import time
from time import sleep
import pandas as pd
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter

class RandomProcedure(Procedure):

    iterations = IntegerParameter('Loop Iterations', default=10)
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='20')
    mode = IntegerParameter('Measurement mode : 2 or 4 probes', default=2)

    DATA_COLUMNS = ['Mode', 'Iteration', 'Timecode', 'Elapsed', 'Random Number']

    def startup(self):
        log.info("Setting the time of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.iterations)
        start = datetime.datetime.now()
        for i in range(self.iterations):
            timecode = dt.now()
            elapsed = (datetime.datetime.now() - start).total_seconds()
            data = {
                'Mode': self.mode,
                'Iteration': i,
                'Timecode': timecode,
                'Elapsed': elapsed,
                'Random Number': random.randint(1, 100)
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break
        if self.mode == 2:   # it s also data[0, 0]
            type_mode = '2_wire_mode'
        elif self.mode == 4:
            type_mode = '4_wire_mode'
        else:
            type_mode = 'undefined'
        print(type_mode)  # i want type_mode to appear in the filename in the queue() function depending on the result here

class MainWindow(ManagedWindow, RandomProcedure):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure,
            inputs=['iterations', 'delay', 'seed', 'mode'],
            displays=['iterations', 'delay', 'seed', 'mode'],
            x_axis='Iteration',
            y_axis='Random Number',
            sequencer=True,  # Added line
            sequencer_inputs=['iterations', 'delay', 'seed', 'mode'],  # Added line
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional
        )
        self.setWindowTitle('GUI Example')
        self.count = 0

    def queue(self, *, procedure=None):
        global type_mode
        self.count = self.count + 1
        filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y")) + '_' + 'mode_' + str(type_mode) + '_' + str(self.count) + '.csv'
        #if put self.mode instead of type_mode it works but always mode 2 printed
        log.info("Constructing the Results with a data file: %s" % filename)
        if procedure is None:
            procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec_())

```

physicsninja, 2020-07-26:
I have opened a new issue for your question, @PYmeas because I don't think if falls under the scope of the issue discuss here, see my response there