pymeasure/pymeasure #78: Missing dependency on pycomedi

ghisvail, 2017-05-29:
Neither the `setuptools` metadata nor the requirements file mention a dependency on `pycomedi`, although the latter is imported in `pymeasure/instruments/comedi.py`. The Sphinx documentation generator is complaining of its absence:
```
/<<PKGBUILDDIR>>/docs/api/instruments/comedi.rst:7: WARNING: autodoc: failed to import module 'pymeasure.instruments.comedi'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/sphinx/ext/autodoc.py", line 529, in import_object
    __import__(self.modname)
  File "/<<PKGBUILDDIR>>/pymeasure/instruments/comedi.py", line 27, in <module>
    from pycomedi.subdevice import StreamingSubdevice
ImportError: No module named 'pycomedi'
```

cjermain, 2017-05-29:
Good eye @ghisvail! The `pycomedi` library is Linux specific, so it should not be required. How about wrapping the import with a try-catch statement?

ghisvail, 2017-05-30:
> wrapping the import with a try-catch statement

Indeed, plus giving some meaningful error message describing why the import failed, i.e. either because the platform is not Linux or because `pycomedi` is missing.

feinsteinben, 2017-05-30:
it can be added as an optional ('extra') requirement in the setup.py file.
http://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-extras-optional-features-with-their-own-dependencies

bilderbuchi, 2020-10-26:
The import guard has been added in b0f08f9c.

closed 2020-10-26 by bilderbuchi