pymeasure/pymeasure #80: Consider installing the tests

ghisvail, 2017-05-29:
Since you decided to embed the test suite inside the package instead of leaving it outside in a common `tests` folder (both of which are totally acceptable and covered by the `pytest` guidelines), why not install the tests with the package? It would give the following benefits:

- Simplify the `setuptools` metadata, by replacing the explicit listing of the modules with a single call to `find_package`,
- Easy testing of the installed package via `python -m pytest --pyargs pymeasure`.

ghisvail, 2017-05-29:
The following change would be enough actually:
```diff
diff --git a/setup.py b/setup.py
index 3426fc4..a57efb2 100644
--- a/setup.py
+++ b/setup.py
@@ -48,6 +48,11 @@ setup(
         'pymeasure.instruments.thorlabs',
         'pymeasure.instruments.tektronix',
         'pymeasure.instruments.yokogawa',
+        'pymeasure/adapters/tests',
+        'pymeasure/experiment/tests',
+        'pymeasure/experiment/tests/data',
+        'pymeasure/instruments/tests',
+        'pymeasure/tests',
     ],
     scripts=[],
     url='https://github.com/ralph-group/pymeasure',
```

cjermain, 2017-05-29:
@ghisvail, that sounds like a useful addition. Can we improve the continuous integration scripts with this?

PR welcome. Also, please add yourself to the authors list.

ghisvail, 2017-05-30:
> Can we improve the continuous integration scripts with this?

The CI setup look fine to me. I would not touch that for now.

What you would get in return is that someone reporting an issue from a venv or user deployment of `pymeasure` could then be asked to run `python -m pytest --pyargs pymeasure` to validate the as-installed package.

PR #81 (2017-06-02): Install the tests alongside the package
Closes #80
closed 2017-06-02 by cjermain