pymeasure/pymeasure #84: Can't exit with Ctrl+C

feinsteinben, 2017-06-09:
Hi,

I'm trying to run a procedure in a safe manner, such that it shuts down all the instruments when closed.
It based on this example: https://github.com/ralph-group/pymeasure/blob/master/examples/Basic/script.py
However, when I try exiting the script with Ctrl+C, it doesn't seem to recognize it. Do you have any suggestions regarding this issue?

Thanks,
Ben.

cjermain, 2017-06-09:
When the code in `script.py` gets to `worker.start()`, any call to Keyboard Interrupt should give you something like the printout below. (I'm using Ctrl+C on Linux.)

```
^C05:16:06 PM: User stopped Worker join prematurely (pymeasure.experiment.workers, WARNING)
05:16:06 PM: Worker has joined (root, INFO)
05:16:06 PM: Stopping the logging (root, INFO)
05:16:06 PM: User stopped Worker execution prematurely (root, WARNING)
05:16:06 PM: Emitting message: b'status' 1 (root, DEBUG)
05:16:06 PM: Finished (root, INFO)
05:16:06 PM: Recorder caught stop command (pymeasure.experiment.listeners, INFO)
```

That first line of the printout is where the interrupt is being caught.

If you insert a Keyboard Interrupt before `worker.start()`, the code is not designed to handle it. You can improve this by simply adding a try-catch statement that catches the interrupt. Since this section is just setting up the experiment, there won't be any instruments in use yet.

@feinsteinben, what OS are you using? The Keyboard Interrupt may be defined differently.

feinsteinben, 2017-06-09:
@cjermain Thanks for your response.
I tried to Ctrl+C after the worker had started.
I'm running that on windows 7.

jrr1984, 2019-06-01:
have you fixed this bug? It seems not to be working the keyboard interrup feature.

cjermain, 2019-06-04:
@jrr1984, can you give more specifics on your observations? What OS are you using? What versions of PyMeasure?

jrr1984, 2019-06-07:
Windows 10 64 bits, using pycharm IDE (i realised this IDE has a bug itself, maybe thats the problem!). The pymeasure version is the latests one from github, tomorrow i can give you more details. I know this is no the place to ask this but do you have any example using the results object when we emit a list as a result like the trace of an spectrometer or an oscilloscope? I would like the Plotter and then the GUI to plot the intensity vs wavelength traces from the spectrometer im using. Thank you very much in advance.

bilderbuchi, 2020-10-26:
@CasperSchippers can you take a look at this?

CasperSchippers, 2021-01-03:
> @CasperSchippers can you take a look at this?

Yes I will, sorry for the late response.

CasperSchippers, 2021-01-24:
The problem seems to be that the wait for the should_stop event (`self._should_stop.wait()` in `StoppableThread.join()`) is not interrupted by the KeyboardInterrupt signal (i.e. ctrl+C). This even seems to be system-specific (as is discussed over [here](https://stackoverflow.com/questions/51083939/why-can-pythons-event-wait-be-interrupted-by-signals-on-some-systems-but-not)). One solution is to break the wait into many very short waits, between which the keyboard-interrupt does do the trick, but I'm looking if other, nicer solutions are possible.

bilderbuchi, 2021-01-24:
Similar sounding, possibly related, Python bug: https://bugs.python.org/issue35935
Could be worthwhile to first try a simpler reproduction procedure, to determine if this is a (the above?) *Python* issue, or something wrong in our code.

CasperSchippers, 2021-01-30:
Yes, it seems very much to be indeed that bug. I've opened a PR to propose a fix (or maybe rather workaround) to this bug.

CasperSchippers, 2021-02-06:
I merged a fix for this problem; it should be resolved right now.

closed 2021-02-06 by CasperSchippers