pymeasure/pymeasure #88: visa write_termination attribute

Laurent15, 2017-07-06:
Hi,

I'm using an equipment controlled by GPIB throught the visa adapter from pymeasure.
For write operation, this equipment is requiering to have an '\n' characters at its end of command.

In the VISAAdapter class, I found that the corresponding argument to used is 'term_chars'.
But, if I'm using it, an exception is raised : 'term_chars' is not a valid attribute for type GPIBInstrument

In fact, the standard visa attribute used to configure this feature is 'write_termination'.

If 'write_termination' is added to the list of arguments allowed, the write_termination characters is taken into account and it's works fine for me.

from pymeasure/adapteer

class VISAAdapter(Adapter):
    """ Adapter class for the VISA library using PyVISA to communicate
    to instruments. Inherit from either class VISAAdapter14 or VISAAdapter15.

    :param resource: VISA resource name that identifies the address
    :param kwargs: Any valid key-word arguments for constructing a PyVISA instrument
    """
    def __init__(self, resourceName, **kwargs):
        ...
        if isinstance(resourceName, int):
            resourceName = "GPIB0::%d::INSTR" % resourceName
        super(VISAAdapter, self).__init__()
        self.resource_name = resourceName
        self.manager = visa.ResourceManager()
        safeKeywords = ['resource_name', 'timeout', 'term_chars',
                        'chunk_size', 'lock', 'delay', 'send_end',
                        'values_format', 'read_termination']

I replace it by :
class VISAAdapter(Adapter):
    """ Adapter class for the VISA library using PyVISA to communicate
    to instruments. Inherit from either class VISAAdapter14 or VISAAdapter15.

    :param resource: VISA resource name that identifies the address
    :param kwargs: Any valid key-word arguments for constructing a PyVISA instrument
    """
    def __init__(self, resourceName, **kwargs):
        ...
        if isinstance(resourceName, int):
            resourceName = "GPIB0::%d::INSTR" % resourceName
        super(VISAAdapter, self).__init__()
        self.resource_name = resourceName
        self.manager = visa.ResourceManager()
        safeKeywords = ['resource_name', 'timeout', 'write_termination',
                        'chunk_size', 'lock', 'delay', 'send_end',
                        'values_format', 'read_termination']

Regards
Laurent

bilderbuchi, 2017-07-06:
Also, I think the VISAAdapter14 or VISAAdapter15 mentioned in the docstring don't exist anymore. 
Maybe this was just and oversight? Possibly connected: #52, 52028601f2dd4f82469579ea45b1864d8984f2ed

cjermain, 2017-07-07:
@Laurent15, thanks for reporting this issue. It looks like this is a leftover from the old PyVISA 1.5 days (see [migration notes to PyVISA 1.5](http://pyvisa.readthedocs.io/en/stable/migrating.html?highlight=term_chars)). Can you make a pull request with this change, and update the doctag to no longer refer to VISAAdapter14 or VISAAdapter15?

bilderbuchi, 2020-05-10:
This has actually been fixed by #171

closed 2020-05-10 by bilderbuchi