pymeasure/pymeasure #89: Procedure example in documentation broken

bilderbuchi, 2017-07-12:
Hi!

When I run the procedure example found at http://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html#running-procedures, nothing is printed in the console, and the `example.csv` file that is created only contains 
```
#Procedure: <__main__.SimpleProcedure>
#Parameters:
#	Loop Iterations: 100
#Data:
Iteration
```
with no data (i.e. counting up from 1 to 100). It seems the worker never joins, as it's just waiting for the timeout (btw. a 1 hour timeout in a demo example is quite the trap ;-) )
This is on pymeasure 0.4.5 installed via pip, and Python 3.6.1 on Windows.

Also trying the other examples does not work, for example when I run the Managed window one, I only get "ValueError: Parameter value is not set", but I think I'll report this in a separate issue.

cjermain, 2017-07-13:
@bilderbuchi, thank you for reporting this! It looks like the examples could use a review. I will do this in the next few days.

bilderbuchi, 2017-07-13:
@cjermain thank you! I have in the meantime tried the example with 0.4.4, and it works as expected. I suspect that in the massive changes by @feinsteinben something broke.

JuMaD, 2017-07-19:
I also get the 'parameter not set' error.

feinsteinben, 2017-07-19:
@bilderbuchi sorry for that :-(
can you copy the example you tried to run?

feinsteinben, 2017-07-19:
I fixed a bug in the results logging (see #95).
After than, both examples (simple example, plotting example) started working.

**OS:** linux, x64.
**Python:** python 3.6.1 (anaconda3 4.4.0), x64.
**PyMeasure**: `latest` (commit `184f50c`), with fix #95.

cjermain, 2017-07-21:
I will make a minor release (PyMeasure 0.4.6) with these once they are tested.

bilderbuchi, 2020-03-05:
This has been fixed in the meantime, closing.

closed 2020-03-05 by bilderbuchi