pymeasure/pymeasure #90: Using the plotter example does not work

bilderbuchi, 2017-07-12:
The [example code](http://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-plotter) has an oversight - SimpleProcedure should be replaced by RandomProcedure.
Even when that's corrected, I don't get any window popping up.

Similar situation for the managed window example (which btw is missing an `import sys`)

This is on pymeasure 0.4.5 installed via pip, pyqtgraph 0.10.0, pyqt 5.6.0, and Python 3.6.1 on Windows.

feinsteinben, 2017-07-19:
I think I fixed the `ValueError: Parameter value is not set` error in the managed window example.
However, now I get another error after putting an experiment in the queue:
```
Process Worker-1:
Traceback (most recent call last):
  File "/home/benfei/.local/share/miniconda3/envs/pymeasure/lib/python3.6/site-packages/zmq/sugar/socket.py", line 356, in send_multipart
    _buffer_type(msg)
TypeError: memoryview: a bytes-like object is required, not 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/benfei/.local/share/miniconda3/envs/pymeasure/lib/python3.6/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/home/benfei/Documents/Github/pymeasure/pymeasure/experiment/workers.py", line 160, in run
    self.update_status(Procedure.RUNNING)
  File "/home/benfei/Documents/Github/pymeasure/pymeasure/experiment/workers.py", line 116, in update_status
    self.emit('status', status)
  File "/home/benfei/Documents/Github/pymeasure/pymeasure/experiment/workers.py", line 96, in emit
    self.publisher.send_multipart(data)
  File "/home/benfei/.local/share/miniconda3/envs/pymeasure/lib/python3.6/site-packages/zmq/sugar/socket.py", line 363, in send_multipart
    i, rmsg,
TypeError: Frame 0 (128) does not support the buffer interface.
```

bilderbuchi, 2017-07-25:
When merging #95 #96 #97, the randomprocedure example works as expected I think. The managed window example works with some caveats, see https://github.com/ralph-group/pymeasure/pull/96#issuecomment-317748308

robmarkcole, 2018-08-30:
I receive the same error as @feinsteinben when running the [gui examples](https://github.com/ralph-group/pymeasure/tree/master/examples/Basic). The GUI launches fine, but I get the error when I hit `queue`

bilderbuchi, 2020-03-05:
The plotter example works by now. The error @feinsteinben @robmarkcole encounter is tracked in #168.

closed 2020-03-05 by bilderbuchi

FlyingCurryMonster, 2022-01-15:
I still get the same error as @robmarkcole.  I had to follow the temporary fix listed in #168  to get the managed window to queue and run the random procedure.

bilderbuchi, 2022-01-21:
This is on what version? Current master? @CasperSchippers any ideas?

CasperSchippers, 2022-01-25:
@FlyingCurryMonster, could you specify what version of pymeasure and python you use, and how you triggered this error?

FlyingCurryMonster, 2022-01-25:
I'm using pymeasure 0.9.0 and python version 3.9.7.  In order for my queued experiments to run, i need to include 
`sys.modules['cloudpickle'] = None`.
Otherwise, my experiment stays queued and will not run and I get this printed out in my console.  

> TypeError: Frame 0 (128) does not support the buffer interface.

Forgive me if this is something I'm missing with pyqt, I'm still fairly new to the library and still just learning how i can add widgets to the main window.

CasperSchippers, 2022-01-26:
I think this is indeed the issue that has been solved by #354, as this fix is not yet in the versions on PyPI or conda-forge (this is yet another reason to have a new version very soon)