pymeasure/pymeasure #91: Multiprocessing behavior reverted and causes error on Unix

ndr37, 2017-07-13:
In process.py, the line:

`set_start_method('spawn', force=True)`
 
was replaced with 

`set_start_method('spawn')`

(currently line 33)
 
which caused it to fail when I ran it on my OSX 10.11.6 with Python 3.4. Restoring the `force=True ` seems to fix the crash.
Is there a reason it was reverted?

ndr37, 2017-07-13:
The stated reason in the most recent version update was:

> 13. Remove the `force-True` argument from `set_start_method('spawn')` in pymeasure/process.py. It doesn't seem to be such an argument.

I don't know what that means. Should it read "doesn't seem to need such and argument"?

bilderbuchi, 2017-07-13:
At least according to the docs, the `force=True` kwarg never existed, and the method itself has only been introduced in Python 3.4 - I guess the classifiers in setup.py should be updated to reflect that (3.4 as a minimum target is pretty reasonable, imho).

ndr37, 2017-07-13:
Just for context, I received a

> RuntimeError('context has already been set')

 error. An internet search for that error leads to at least a bug report that reports a similar problem and notes adding `force=True` as a kwarg fixed the error :

http://grokbase.com/t/python/python-bugs-list/148q63vwrz/issue22255-multiprocessing-freeze-support-raises-runtimeerror

which is what lead me to noticing that it had been removed. I can tell you it does definitely fix the problem, at least superficially.

cjermain, 2017-07-13:
I thought I had tested this correctly, but as I am re-running the examples I see this same issue on Linux. This is certainly a regression, and I will make a new release on fixing these issues.

@bilderbuchi, if I recall correctly I found this fix online. I was surprised not to find it in the docs, but it clearly is there in practice.

@ndr37, can you update this line? Thanks for reporting it!

This shows that PyMeasure needs a bit more unit testing. The current tests do not have enough coverage. I will tackle some of this as time allows.

bilderbuchi, 2017-07-13:
I did a quick check  of this today and yes, the coverage is at approximately 30% of the codebase.

ndr37, 2017-07-13:
Line has been reverted

closed 2017-07-13 by ndr37

bilderbuchi, 2017-07-13:
Btw, the docs [say](https://docs.python.org/3.6/library/multiprocessing.html#contexts-and-start-methods):

> set_start_method() should not be used more than once in the program.
> 
> Alternatively, you can use get_context() to obtain a context object. Context objects have the same API as the multiprocessing module, and allow one to use multiple start methods in the same program.
> ...
> A library which wants to use a particular start method should probably use get_context() to avoid interfering with the choice of the library user.

so maybe get_context would be more appropriate to get around the "context already set" problem? I don't know the technical background, though.