pymeasure/pymeasure #92: It's not possible to test locally installed pymeasure

bilderbuchi, 2017-07-13:
Today I ran `py.test` on my locally installed pymeasure. This breaks with 
```
______________ ERROR collecting experiment/tests/test_workers.py ______________
..\_pytest\config.py:1213: in getoption
    val = getattr(self.option, name)
E   AttributeError: 'CmdOptions' object has no attribute '--runslow'

During handling of the above exception, another exception occurred:
experiment\tests\test_workers.py:39: in <module>
    not pytest.config.getoption("--runslow"),
..\_pytest\config.py:1223: in getoption
    raise ValueError("no option named %r" % (name,))
E   ValueError: no option named '--runslow'
!!!!!!!!!!!!!!!!!!! Interrupted: 1 errors during collection !!!!!!!!!!!!!!!!!!!
```
because the `conf.py` file is not packaged along with the tests. This makes the `getoption` bail because the option has not been added in the conf.py.

This is not visible on CI because you run the tests on the local pymeasure repo directory, instead of an installed version, which might be a better idea because it reflects your users' situation better. That should probably be changed.

The easiest fix for the present issue that I could see, since you apparently decided to use the test structure which [distributes tests with your application](https://docs.pytest.org/en/latest/goodpractices.html#tests-as-part-of-application-code), is to move the conftest.py into the `pymeasure` directory to distribute it with the rest of the tests. 
It's still possible to test a locally available, not installed, repo clone by calling pytest from the `./pymeasure` directory, not the repo root dir.

cjermain, 2017-07-18:
@bilderbuchi, I can reproduce this. I am working on a branch to overhaul the tests and will address this.

bilderbuchi, 2017-07-18:
I know that finding the proper test structure do cover all usecases can be really tricky.