pymeasure/pymeasure #94: Introduce a compatibility layer to enable pymeasure to use InstrumentKit drivers

bilderbuchi, 2017-07-13:
A "compatibility layer" that would enable pymeasure to use existing IK drivers would be a great first step to move the pymeasure driver portion to InstrumentKit, and use that package's drivers, If necessary, incompatible information (e.g. units) could be discarded in a first iteration, to reduce the complexity of this endeavour.
cc @cjermain @scasagrande #53

cjermain, 2017-07-21:
@bilderbuchi, PR welcome. I plan to focus my time on better test coverage at this point, but would be happy to see this effort progress.

bilderbuchi, 2017-07-21:
Better test coverage is good! It will definitely take me quite some time to get a feel for both codebases enough so that I can try to contribute something.

bmoneke, 2022-06-20:
I started a first version in my branch [IKcompatibility](https://github.com/bmoneke/pymeasure/tree/IKcompatibility). It uses the testing environment, we are implementing right now (branch [pymeasure/protocol-tests](https://github.com/pymeasure/pymeasure/tree/protocol-tests), therefore I did not do a pull request.

At least using the ProtocolAdapter works as expected. I did not yet connect a real instrument.

bmoneke, 2022-07-01:
Looking more into the GUI of pymeasure, I did not find anything, which required a specific type of instrument.

Does an instrument require anything special, in order to be able to be used by pymeasure GUI? @CasperSchippers @msmttchr

If not, it could be advertised, that any python instrument may be used, increasing the coverage of pymeasure GUI.

msmttchr, 2022-07-01:
> Looking more into the GUI of pymeasure, I did not find anything, which required a specific type of instrument.
>
> Does an instrument require anything special, in order to be able to be used by [...]

I think this is correct. The GUI provide a framework to execute experiments and data is usually retrieved from Pymeasure instrument, but this not strictly required.

neal-kepler, 2022-07-01:
> Looking more into the GUI of pymeasure, I did not find anything, which required a specific type of instrument.
>
> Does an instrument require anything special, in order to be able to be used by [...]

I second @msmttchr's assessment. The gui doesn't care how you get your data, only that you can fit your code into the startup, execute, and shutdown methods of the procedure class and that you emit the data in a format compatible with your chosen formatter.

bmoneke, 2022-07-01:
Thanks for corroborating my first assessment.

As python itself is already the "compatibility layer", I propose to close this issue.

At least I do not see any benefit in writing "compatibility layers" as you have to look in another library anyway in order to understand that specific instrument.

bilderbuchi, 2022-07-02:
Hm, that is actually a good point! It would be cool to add some documentation and an example demonstrating how to, e.g., set up an experiment and GUI run with an IK instrument driver. Then I think we can indeed close this.

bilderbuchi, 2022-07-13:
Thank you, closed by #647.

closed 2022-07-13 by bilderbuchi

scasagrande, 2022-07-13:
nice!