pymeasure/pymeasure #102: Improve documentation

bilderbuchi, 2017-07-20:
Migrated from TODO

cjermain, 2017-07-21:
PRs welcome!

bilderbuchi, 2017-07-23:
Doctests in both pymeasure code and runnable examples in the documentation would be useful, see [here](https://github.com/ralph-group/pymeasure/issues/98#issuecomment-317273099).

bilderbuchi, 2017-07-28:
I found this a good, short, read about what constitutes good documentation: https://www.divio.com/en/blog/documentation/

bilderbuchi, 2017-08-29:
Some doctests executing code in the documentation has been added in #111, so this can be used as a guide for adding documentation doctests elsewhere. 

I have also looked a bit into adding doctests to the pymeasure code docstrings, but since most things are very interconnected and often depend on GUI features, I did not find feasible/sensible opportunities.

bilderbuchi, 2021-12-03:
document that
* [ ] `__init__` should be at the top of the class
* [ ] common pattern: bool properties with map_values
* [ ] common pattern: Channels for instruments
* [ ] common pattern: inheritance between instrument models -> nice example AnritsuMS9740A!
* [ ] PR single topic (1 instrument), small size makes reviews much easier

bilderbuchi, 2022-01-13:
* [ ] Maybe we should put some recommendation how to deal with atypical communication needs of some instruments (and pointer to examples) in the docs?  One [c|sh]ould encapsulate the device-specific comms requirements in a custom Adapter derived e.g. from VISAAdapter, instead of overriding the methods of `Instrument`. See e.g. #352/ [TopticaAdapter](https://github.com/dkriegner/pymeasure/blob/3d5f57b4d8209dfa2ae32f8794eb94e1f36fa794/pymeasure/instruments/toptica/adapters.py#L31), there are also other cases of this.  Cf .#357

LongnoseRob, 2022-01-13:
> * [ ]  Maybe we should put some recommendation how to deal with atypical communication needs of some instruments (and pointer to examples) in the docs?  One [c|sh]ould encapsulate the [...]

Very good idea, maybe we could extend this into a kind of best-practice-section

bilderbuchi, 2022-01-13:
Yeah, if we re-organize the docs according to the scheme linked above ( tutorials, how-to guides, explanation, reference), this should naturally fall out of that.

bilderbuchi, 2022-01-24:
also see #364.

Instrument naming patterns:
One pattern you can use for the adapting the names, while retaining full flexibility, is: 

```
class BaseClass:
    def __init__(self, adapter, name=None, **kwargs):
        #...
        super().__init__(
            adapter,
            name or "Base instrument name",
            **kwargs
        )

class DerivedClass(BaseClass):
    # or do the name handling like this
    def __init__(self, adapter, name="Derived instrument name", **kwargs):
        #...
        super().__init__(
            adapter,
            name,
            **kwargs
        )
```
This will also enable you to give your instanced instruments setup-specific names.

bilderbuchi, 2022-05-19:
* [ ] Extend https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#writing-properties to explain more about the different options to use, and how to use them (e.g. `check_set|get_errors`) to avoid what kind of problems.

bilderbuchi, 2022-08-05:
* [ ] Good naming scheme for boolean property - able to formulate as a question (` output_enabled`)
* [ ] "Actions"/verbs should typically be methods, not properties (`recall()`, `trigger_scan()`)

PR #712 (2022-11-06): Update and extend documentation
This should reduce confusing situations like [this](https://github.com/pymeasure/pymeasure/discussions/710#discussioncomment-3905231).

Closes #102.
closed 2022-11-06 by bilderbuchi