pymeasure/pymeasure #98: Add tests for all non-instrument features

bilderbuchi, 2017-07-20:
Migrated from TODO

cjermain, 2017-07-21:
This is currently a top priority for the next major release, PyMeasure 0.5. I plan to do some reorganization of the test, but would welcome any contributed PRs to improve the test coverage.

bilderbuchi, 2017-07-23:
There is a connection to #102 here, as the doctest facilities of pytest could be used to test some of the examples in the docs. Also, doctests in the pymeasure code docstrings could be a good way to get some coverage increases going, while at the same time improving documentation.
Pytest can [consume](https://docs.pytest.org/en/latest/doctest.html) doctests, both in .py and .rst files. Unfortunately, some of the features of Sphinx doctests [are not supported yet](https://github.com/pytest-dev/pytest/issues/758) by pytest, so more complex examples will probably not be reasonably testable.
Sphinx doctest functionality [is quite featureful](http://www.sphinx-doc.org/en/stable/ext/doctest.html), but to keep compatibility with pytest as much as possible, [the simple doctest format](http://www.sphinx-doc.org/en/stable/ext/doctest.html#confval-doctest_test_doctest_blocks), i.e.  separate rst pararagraphs starting with `>>>`, are preferable.

These are more or less research notes, I'll try to poke at seeing if we can integrate what's in the docs already if I find the sparetime during the summer.

bilderbuchi, 2017-08-29:
doctests for the documentation have been added in #111.