pymeasure/pymeasure #105: Leaked semaphores?

bilderbuchi, 2017-07-27:
When running the managed window example in Spyder, closing the window, and running it again, I get a crash of the consol kernel. This does not happen in pycharm. @feinsteinben can also reproduce this.
Sometimes I noticed this error message after closing for the first time:
```
/home/bilderbuchi/anaconda3/envs/pymeasure/lib/python3.6/multiprocessing/semaphore_tracker.py:129: UserWarning: semaphore_tracker: There appear to be 28 leaked semaphores to clean up at shutdown
```
and some userwarning about how the application should be shutdown with ctrl-q and some other key combos.

cjermain, 2017-07-31:
I am also getting this after merging #95, #96, #97.

```
Traceback (most recent call last):
  File "/home/colin/.miniconda3/envs/pymeasure/lib/python3.5/multiprocessing/util.py", line 253, in _run_finalizers
    finalizer()
  File "/home/colin/.miniconda3/envs/pymeasure/lib/python3.5/multiprocessing/util.py", line 185, in __call__
    res = self._callback(*self._args, **self._kwargs)
  File "/home/colin/.miniconda3/envs/pymeasure/lib/python3.5/multiprocessing/synchronize.py", line 87, in _cleanup
    from .semaphore_tracker import unregister
SystemError: Parent module 'multiprocessing' not loaded, cannot perform relative import
```

cjermain, 2017-08-06:
This error goes away using `fork` instead of `spawn`, but it's not clear yet whether this is a Python or PyMeasure bug.

cjermain, 2017-08-12:
This looks like a CPython bug, but have not gotten to the bottom of it completely. It only occurs on Unix, and only with `spawn`.

bilderbuchi, 2021-03-04:
I can't reproduce this anymore, on Ubuntu, python 3.9, Spyder 4.2.3, so I'll close this. Feel free to re-open with a precise repro procedure should you encounter this again.

closed 2021-03-04 by bilderbuchi