pymeasure/pymeasure #107: All Threads by version 0.5

cjermain, 2017-08-12:
I'm considering the idea of switching `Worker` objects to run in threads instead of processes, which would largely eliminate the need for multiprocessing in PyMeasure. I would love everyone’s feedback on this before making the final decision. Below I've outline a number of reasons why I think this is a good idea.

1. The original reason to move to `multiprocessing.Process` was to deal with a performance bottleneck that was actually solved a number of releases ago (not enough `sleep` was keeping the CPU occupied).  (The Global Interpreter Lock (GIL) restricts Threads to one CPU.) While it is nice to leverage a second CPU, it should not be necessary at this point.

2. Since Windows does not support forking a process, this forces us to use spawn. We have been using a hack (`set_start_method('spawn', force=True)`, see [line 33 in process.py](https://github.com/ralph-group/pymeasure/blob/184f50cd1d938dd4ace84db8747ad9ca8fe814cb/pymeasure/process.py#L33)) to get this working on Linux/MacOS. The `force=True` part is not even in the official docs, so overall this seems like a bad idea to rely on this method.

3. Pickeling procedure objects is required to send them to a new Process. In the case when the child Procedure class is defined in a file loaded dynamically, this becomes a significant issue (for spawned processes) and the new Process does not properly import the Procedure class. This is arguably a limitation of CPython currently. I created a work-around solution that re-imports the procedure class when unpickleing, but this is overly complex for what we are trying to do.

Going back to Threads eliminates these issues because there is no creation of a new Process, and thread-safe shared resources can be easily accessed. Overall, I think this will greatly simplify maintenance of the package, the unit testing, and will not significantly affecting the performance. A user who needs to access the other CPUs for a specific task can always launch their own sub-processes from the Worker.

My goal is to make this transition by the next major version 0.5. I'm interested in your thoughts.

CC: @ndr37, @guenp, @minhhaiphys, @bilderbuchi, @feinsteinben, @melund

bilderbuchi, 2017-08-12:
My gut feeling is that for the tasks that pymeasure needs to do, you're most of the time I/O bound (waiting for instruments) not CPU bound, so maybe a simpler approach like concurrent futures and friends would be the way to go? 
Regarding the pickling, I don't know the details of the problems you have encountered, but this sounds like interprocess communication needs to pass hashable objects, and your procedure isn't one (dicts instead of ordereddicts inside could be a reason).
Finally, as a closer interop with instrumentkit is intended, probably it's worthwhile to either check their code or, better, ask them how they have solved this situation. @scasagrande, any thoughts on the OP?

cjermain, 2017-08-17:
@bilderbuchi, agreed that I/O time is the significant time constraint here. The pickeling issue is actually on reconstructing a class that was dynamically imported (in a spawn process it won't actually have the class available at process construction).

Any other thoughts?

Laogeodritt, 2017-11-16:
Is the `Plotter` intended to still use multiprocessing, now that the `Worker` is just threaded? Or was this an oversight?

Noticed this discrepancy while handling #122.

cjermain, 2017-11-20:
@Laogeodritt, the `Plotter` should be switched to threading. This was an oversight on my part.

See @dvspirito's comment in #109.