pymeasure/pymeasure #109: Forking/Pickling issue when worker.start() is called?!

JuMaD, 2017-08-25:
Using the ManagedWindow class, I can add experiments to the queue. However, once they should "start" (i.e. the next() method is used) the whole .py file gets interpreted again instead.

In the debugger I traced it down to a new thread being started when the worker is started. When Manager.next._worker.start() is called). Due to the error message below my best guess is, that it has something to do with the forking/pickling issue(s). Since I have no clue about multi-threading, I hope someone can help me fix this. 

[image]

cjermain, 2017-08-25:
@JuMaD, thanks for reporting this issue!

Can you provide more context? Which OS are you using? Are you running one of the examples?

JuMaD, 2017-08-26:
I'm working on Windows 7/10. The problem quite literally popped-up in an own program, but I tested it with the ManagedWindow example (which has a typo in the docs btw ;) ) where it is also an issue.

cjermain, 2017-08-27:
Which version of PyMeasure are you running? Version 0.4.5 had some regressions concerning the Worker behavior that we didn't catch before the release.

JuMaD, 2017-08-28:
I'm running the latest version with some minor adjustments (just expanded the GUI of ManagedWindow with a list to pick an instrument and starting the queue only on Button click [Disabled start_on_add and connected the button to "resume"]).

Everything else is running just fine. But as I said: When I debug, I can see that "starting" the worker leads to a new thread instead.

bilderbuchi, 2017-08-28:
It's probably helpful for the maintainers if you paste the code in question, or a minimum reproducible example and the output you get, here instead of making them search the changes you could mean. ;-)

JuMaD, 2017-08-28:
Ok then. Here's the (adjusted) example from the docs

`
print('Running')
class RandomProcedure(Procedure):
    iterations = IntegerParameter('Loop Iterations')
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='12345')

    DATA_COLUMNS = ['Iteration', 'Random Number']

    def startup(self):
        log.info("Setting the seed of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.iterations)
        for i in range(self.iterations):
            data = {
                'Iteration': i,
                'Random Number': random.random()
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break

class MainWindow(ManagedWindow):
    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure,
            inputs=['iterations', 'delay', 'seed'],
            displays=['iterations', 'delay', 'seed'],
            x_axis='Iteration',
            y_axis='Random Number'
        )
        self.setWindowTitle('GUI Example')

    def queue(self):
        filename = 'test.csv'

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)`

And my edited ManagedWindow class:
`class ManagedWindow(QtGui.QMainWindow):
    """ The ManagedWindow uses a Manager to control Workers in a Queue,
    and provides a simple interface. The queue method must be overwritten
    by a child class which is required to pass an Experiment containing the
    Results and Procedure to self.manager.queue.
    """
    EDITOR = 'gedit'

    def __init__(self, procedure_class, inputs=(), displays=(), x_axis=None, y_axis=None,
                 log_channel='', log_level=logging.INFO, parent=None):
        super().__init__(parent)
        app = QtCore.QCoreApplication.instance()
        app.aboutToQuit.connect(self.quit)
        self.procedure_class = procedure_class
        self.inputs = inputs
        self.displays = displays
        self.log = logging.getLogger(log_channel)
        self.log_level = log_level
        log.setLevel(log_level)
        self.log.setLevel(log_level)
[... 297 lines cut ...]
            self.abort_button.setEnabled(False)

    def queued(self, experiment):
        self.abort_button.setEnabled(True)
        self.browser_widget.show_button.setEnabled(True)
        self.browser_widget.hide_button.setEnabled(True)
        self.browser_widget.clear_button.setEnabled(True)

    def running(self, experiment):
        self.browser_widget.clear_button.setEnabled(False)

    def abort_returned(self, experiment):
        if self.manager.experiments.has_next():
            self.abort_button.setText("Resume")
            self.abort_button.setEnabled(True)
        else:
            self.browser_widget.clear_button.setEnabled(True)

    def finished(self, experiment):
        if not self.manager.experiments.has_next():
            self.abort_button.setEnabled(False)
            self.browser_widget.clear_button.setEnabled(True)
`

Everything else is the same as in 0.4.6. Starting the worker (i.e. in this example clicking "queue") leads to a re-interpretation of the whole code and a new window in a new thread instead of a running "experiment". What I mean by this is that even 'print('Running')' gets executed a second time.

cjermain, 2017-08-28:
@JuMaD, I understand. The fact that `print('Running')` is called twice is actually expected. When the Worker starts, it produces a spawned process on Windows. This spawned process starts from scratch, so it must re-import everything and re-establish the procedure class. In doing so, it will re-run the code outside of the procedure class when importing it (since it is not hidden behind a `__main__` if clause), just like this occurs the 1st time when you import it in the main process.

I suspect that you are having an error in the spawned process. I would recommend using `tblib` to help diagnose any errors in the new process. You can add the following code to the main `__init__.py` file in pymeasure:

```python
import tblib.pickling_support
tblib.pickling_support.install()
```
This will ensure that your traceback is properly shown.

Overall, this issue seems to stem from the multiprocess behavior. I recently suggested moving back to all threads (#107), which I suspect would fix this issue. In my mind this is another example where the ability to use multiple CPUs with multiple processes is outweighed by the complexity of troubleshooting edge cases like this one.

FYI, try using ``` (three `) for multi-line code in Markdown.

Does that make sense?

JuMaD, 2017-08-28:
@cjermain, thank you for your tips. 

> I suspect that you are having an error in the spawned process. 

--> That is what I suspected, too. My problem is that I basically have no clue about multi - processing / threading whatsoever, i.e. troubleshooting is difficult for me.

Since I get a new ManagedWindow when the worker is called I suspected that even the `if __name__ == __main__` protected statements, namely
```
app = QtGui.QApplication(sys.argv)  
    window = MainWindow()
    window.show()   
    sys.exit(app.exec_())
```
get called again. So I added a `print("inside __main__")` and a `print("outside __main__")` inside and outside the protected area. This is what I get:

```
outside __main__
inside __main__
outside __main__
outside __main__
inside __main__
```
In the debugger I can see that I ended up with three processes: two times "MainThread" and one "Dummy-6". If I understood you correctly, this explains the obove output (first __main__:1xoutside&1x inside, Dummy-6: outside, second __main__: outside&inside)

I copy&pasted the suggested code - do I have to do anything else? Or where do I see the traceback?

cjermain, 2017-08-29:
@JuMaD, this is an interesting problem. I would have expected only 2 processes to be running at a single time. You won't see the traceback unless there is an explicit error. I'll think about this further.

JuMaD, 2017-08-29:
Thank you for the quick reply and the assistance! 

The second "MainThread" is what confused me, too. 
As for the new window: Could it be, that the problem is, that ManagedWindow is calling  QtCore.QCoreApplication.instance() outside the '__main__' protection?

cjermain, 2017-08-31:
Do you have a link to a fork with the changes you made?

JuMaD, 2017-08-31:
Of course: https://github.com/JuMaD/pymeasure

JuMaD, 2017-09-15:
Have you found the time to have a closer look at this issue?

cjermain, 2017-09-16:
@JuMaD, I have not had the opportunity. I will make some time tomorrow to check it out.

cjermain, 2017-09-17:
@JuMaD, this is most likely an issue with spawn. I can not reproduce it on Linux with your branch `extend_managedwindow` -- on Windows it would use spawn to launch the process. I would suggest that I make the multiprocessing to thread switch (#107) instead of debugging further, so that we are no longer using spawn. If that doesn't remove your issue, it should make it much easier to debug.

You mentioned a typo in the docs. Can you make a PR fix?

JuMaD, 2017-09-17:
Thanks a lot for checking it out! And I agree - switching to thread should "solve" the issue on my end (and propably all other windows users)

I think it was just one typo - I'll check it in more detail tomorrow.

ndr37, 2017-09-18:
I'm not sure if this will be helpful, but I encountered this problem in version 4.6 as well on Windows 7 and, as far as I can tell, reverting to version 4.4. (I did not check version 4.5 because there isn't a wheel).

JuMaD, 2017-10-10:
Thanks ndr37! Reverting back to 4.4 fixes the 'new Window' issue. However, I forgot to make a branch after forking and now I have no clue how to push (only) my changes into the branch with version 0.4.4 :(

Is it worth for me to figure out a way to cherry-pick all commits minus the ones that change to multiprocessing or will the multiprocessing to thread switch be done any time soon?

cjermain, 2017-10-10:
@JuMaD, the multiprocessing to thread switch #113 is complete. If you are willing to test it I would greatly appreciate it. I've been meaning to write more unit tests before merging, but it should work correctly as is.

JuMaD, 2017-10-10:
Great! Is it dev/threads?

cjermain, 2017-10-10:
Yes, its in `dev/threads`.

dvspirito, 2017-10-23:
I had the same issue with `Plotter`, after updating to version 0.5.
This https://github.com/dvspirito/pymeasure/commit/b64a83d128055489b430074a95cc4d054ba1e52e fixed it, however I didn't test much (I'm on Win7). I don't know if this fix can have side effects on other classes.

cjermain, 2017-11-13:
@dvspirito, thanks for bringing this up. I left `Plotter` on multiprocessing when I did the switch (forgetting about the Windows-specific bug) -- it should have also been switched over. Feel free to make a PR with your fix and I will merge.

bilderbuchi, 2020-03-06:
I think this has been fixed in the meantime.

closed 2020-03-06 by bilderbuchi