pymeasure/pymeasure #114: Tests for 32 bit builds

cjermain, 2017-09-19:
Currently we are running unit tests for:

- Windows 32 bit: Python 3.4, 3.5, and 3.6
- Windows 64 bit: Python 3.4, 3.5, and 3.6
- Linux 64 bit: Python 3.4, 3.5, and 3.6
- macOS 64 bit: Python 3.4, 3.5, and 3.6

I suggest that we drop support for 32 bit Windows to reduce the test run time. 

Any thoughts?

bilderbuchi, 2017-09-19:
yeah why not. Are the windows builds running that much longer than the Travis builds?

cjermain, 2017-09-25:
@bilderbuchi, great question. The travis builds are doing both Linux and macOS. I was thinking to break them up, and use CircleCI to run the Linux builds.

bilderbuchi, 2017-09-26:
With the amount of activity this report is generally seeing, is it worth the hassle of supporting 3 different CI configurations to shave a single-digit number (?) of minutes from the testing time? Or do you expect the amount of testing, both in length and frequency, to increase substantially in the future?

bilderbuchi, 2017-09-26:
Report -> repo, thank you phone...

cjermain, 2017-09-26:
@bilderbuchi, that is a good point. Overall, I want to make sure that the CI does not get in the way, and that unit testing is quick. For now, lets leave it as is and keep this issue open if people have other thoughts.

cjermain, 2017-10-19:
It looks like the main delay for the CI tests are actually the macOS builds. TravisCI has a fairly consistent backlog of macOS builds in their status (https://www.traviscistatus.com/).

closed 2021-02-16 by bilderbuchi