pymeasure/pymeasure #116: Unexpected behaviour of Plotter and GUI window in examples (and modified ones)

aagentile, 2017-10-17:
(as the first let me thank you for the nice effort leading to this library!)

I am having some troubles with reproducing the examples as in
[http://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html](url)
(which might be an old issue)
as well in modifying the example structures to monitor actual serial instruments.

- Configuration: PyMeasure-0.4.6 running on Conda installed 3.5.2 Python 64bit on Windows 7 platform. 

- [ ] Reproducing the examples
If I attempt to reproduce the examples fully in a Jupyter Notebook , I get no error message, but the datafile will not update with new "random" data, and the process will get stuck (waiting for the worker to join as in #89 ?) waiting for timeout. 
I found a workaround by running definitions in a separate script called via the notebook, and then running in the notebook only the last part. I cannot explain why.

- [ ] Passing serial handles to procedure breaks the code
This is a bit more tricky. I was trying to replace the "Random" procedure with temperature readings from a Lakeshore331. Tested via PyMeasure in advance, it works providing floats as a response. I therefore modified the Procedure class in the simple way as follows to fit my scope:

`#This does not work

     class LS334_Procedure(Procedure):

        iterations = IntegerParameter('Loop Iterations')
        delay = FloatParameter('Delay Time', units='s', default=3)

        DATA_COLUMNS = ['Iteration', 'Sample_T']
        
        def __init__(self, LakeShore = None):
            self.status = Procedure.QUEUED
            
            self.LakeShore = LakeShore
            self._update_parameters()
            self.gen_measurement()

        def startup(self):
            print(self.LakeShore.instrid)

        def execute(self):
            print("Starting the loop of %d iterations" % self.iterations)
            for i in range(self.iterations):
                data = {
                    'Iteration': i,
                    'Sample_T': self.LakeShore.temperature_A
                }
                self.emit('results', data)
                print("Emitting results: %s" % data)
                sleep(self.delay)
                if self.should_stop():
                    print("Caught the stop flag in the procedure")
                    break

where the initialisation parameter "LakeShore" is the usual handle provided by
`adapter = LakeShoreUSBAdapter("COMxx")
from path/to/instruments.lakeshore.lakeshore331 import LakeShore331
LakeShore = LakeShore331(adapter)`
in the Jupyter notebook,

Interestingly, the first time I run it from a wiped kernel, the code goes without warnings nor errors into the same "frozen" mode of issue #89 until timeout occurs. GUI/Plotter windows correctly display, but no data are logged on file nor on screen.

If I launch it again I get instead in Plotter the error message
> TypeError : can't pickle traceback objects
()
whereas the GUI window crashes.

`#an ideally equivalent code works
    
   class NewProcedure(Procedure):

        iterations = IntegerParameter('Loop Iterations')
        delay = FloatParameter('Delay Time', units='s', default=0.2)

        DATA_COLUMNS = ['Iteration', 'Chosen Number']
        
        def __init__(self, seed = 10):
            self.status = Procedure.QUEUED
            self.seed = seed
            self._update_parameters()
            self.gen_measurement()

        def startup(self):
            log.info("Setting the seed of the random number generator")
            random.seed(self.seed)

        def execute(self):
            log.info("Starting the loop of %d iterations" % self.iterations)
            for i in range(self.iterations):
                data = {
                    'Iteration': i,
                    'Chosen Number': self.seed 
                }
                self.emit('results', data)
                log.debug("Emitting results: %s" % data)
                sleep(self.delay)
                if self.should_stop():
                    log.warning("Caught the stop flag in the procedure")
                    break`

In the Plotter, I just call both of them with

`if __name__ == "__main__":
    console_log(log)

    log.info("Constructing a RandomProcedure")
    procedure = gpl.LS334_Procedure(LakeShore = 30)
    ...`

I am left clueless now, am I doing something intrinsically wrong without noticing, or is there an issue with data logging and emitting from instrumentation? Any clever hint?
An example of how to use an instrument with a Procedure would be very useful in the documentation!

physicsninja, 2017-10-17:
We are currently in the process of switching away from processes to threads. @cjermain can speak to this more, but I think this problem is due to the intricacy of implementing PyMeasure using processes. 

If you want a temporary fix until that is sorted out, I suggest reverting to version 0.4.4 (`pip install pymeasure=0.4.4`), it seems some of the massive upgrades going from 0.4.4 to 0.4.5-6 broke some things. That should make it throw the 

`TypeError : can't pickle traceback objects
()
whereas the GUI window crashes.`

on the first try. 

The reason for this error is because the procedure has to be pickle-able, instances of Pymeasure instruments are evidently not (@cjermain correct me if I'm wrong). To fix this, one creates the instance itself in the procedure, typically during the startup routine.  In your example that would look like

class LS334_Procedure(Procedure):

    iterations = IntegerParameter('Loop Iterations')
    delay = FloatParameter('Delay Time', units='s', default=3)

    DATA_COLUMNS = ['Iteration', 'Sample_T']
    
    def __init__(self, LakeShoreAddress):
        self.status = Procedure.QUEUED
        
        self.LakeShore = LakeShore(LakeShoreAddress)
        self._update_parameters()
        self.gen_measurement()

    def startup(self):
        print(self.LakeShore.instrid)

    def execute(self):
        print("Starting the loop of %d iterations" % self.iterations)
        for i in range(self.iterations):
            data = {
                'Iteration': i,
                'Sample_T': self.LakeShore.temperature_A
            }
            self.emit('results', data)
            print("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                print("Caught the stop flag in the procedure")
                break

A more full example of using instruments if found in the I-V measurements section of the [examples ](https://github.com/ralph-group/pymeasure/tree/master/examples/Current-Voltage%20Measurements)

Unfortunately, this may result in a type of the same problem you experienced where running it twice in a row will cause the second instance to fail. The reason is that if the process containing the instance of the instance of the serial instrument isn't closed appropriately then it will tie up the serial port and Windows doesn't allow multiple access of serial ports. Terminating the spurious lingering process should solve that problem if it arises.

JuMaD, 2017-10-17:
The error is most likely due to the known issue with multiprocessing (compare #109). This will be solved once dev/threads is pulled. For now you could fork pymeasure and merge the branch (works just fine with me} or revert back to an older version (easier if you don't want to have to update your local copy manually.

The example with randomprocedure works fine for me with both the plotter as well as managed window. However, I do agree an example using the dummy instrument would be useful. I'm also new to 'all this' but the main difficulty seems to be getting data from instruments as it's handled very differently by different instruments, you have to wait for buffers etc.
Maybe try using either visual or auditory output of your instrument within a procedure to test that everything is working (e.g. A beeper that plays a triad from a given start frequency,...)

cjermain, 2017-10-18:
@aagentile, thanks for reporting this! And thanks @physicsninja and @JuMaD for the quick responses. 

Both of the issues appear to be multiprocessing related, and are currently Windows specific (PyMeasure 0.4.6). I will push a new release out now after merging `dev/threads`, so people can start using that. I was originally planning to add more unit tests, but I don't think its worth waiting on.

With regards to the instrument creation, the typical workflow is to create and setup your instruments in the `startup` method. If anything fails in either the `startup` or `execute` methods, the `shutdown` method will always be called. That way you can clean up safely (like turning the power off), if the program has an issue.

aagentile, 2017-10-19:
Thanks all of you for the quick replies!

Understood the issue, thanks for explaining me the problem.
So far I managed a quick fix using @physicsninja suggestion. True, it crashes when called twice in a row, yet this was no major issue. 

I will look forward the new release if close in time, otherwise I will do a merge as suggested by @JuMaD . 
Also thanks for the hint about the startup method, had not yet spotted it and I might have re-invented the wheel myself.

I will keep updated on the latest news :)

JuMaD, 2017-10-19:
Actually @cjermain  just released the new version. Just update and you will be fine. If you already added pip as PATH to your environment variables just type this into your console:
`sudo pip install pymeasure --upgrade`

If you have not here's how to do it: Go Environment Variables, Click "New" and insert PATH as name and "<path>\Scripts" as value where <path> is the path of your Anaconda installation.

bilderbuchi, 2017-10-19:
I thought it's not recommended anymore to `sudo pip` anything, and instead just `pip install`, which installs into your home directory somewhere? I have successfully been using that method up until I switched to Anaconda on Linux, too..

bilderbuchi, 2020-03-06:
I think this has been fixed in 0.5, feel free to reopen if that's not correct.

closed 2020-03-06 by bilderbuchi