pymeasure/pymeasure #120: Logarithmic and semilog plotting using ManagedWindow

Laogeodritt, 2017-11-05:
Is there presently any way of showing log-log or semi-log plots using ManagedWindow? Looks like [pyqtgraph supports it](http://www.pyqtgraph.org/documentation/graphicsItems/plotitem.html#pyqtgraph.PlotItem.setLogMode) but I'm not seeing those facilities called in PyMeasure at the moment.

Main use case for this would be for multi-decade frequency sweeps (x-axis) and plotting power data logarithmically when we're obtaining linear units from an instrument.

cjermain, 2017-11-07:
@Laogeodritt, great question! I believe that you can right click on the plot to open up the pyqtgraph menu. There should be an option to adjust the plot and axes. If not, PR is certainly welcome.

Laogeodritt, 2017-11-07:
Yes, that does work! Thanks, I didn't realise there was a context menu.

Do you know if there's any API support for setting these programmatically in PyMeasure right now? I can see that pyqtgraph does but I don't think the possibility is forwarded to the PyMeasure GUI.

If not, maybe something I can work on later. Since we're talking experiment-specific GUIs and automation, having the display automatically configured would certainly be a useful feature.

cjermain, 2017-11-12:
I'm glad that works. You can certainly adjust the plot programmatically. I would suggest checking out [curves.py](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/display/curves.py) and the pyqtgraph docs. I certainly welcome enhancing the PyMeasure support for different axes.

PR #126 (2017-11-20): Exposing pyqtgraph.PlotItem API for plot display customisation
Fixes #120. These changes expose convenient, documented means for a `pymeasure` user to customise the plot display, e.g. using logarithmic axes, setting axis ranges, etc.

This pull request introduces new APIs, and does not introduce breaking changes.

* `ManagedWindow.plot` and `PlotterWindow.plot` attributes are now documented, public APIs.
* Added `ManagedWindow.setup_plot()` and `Plotter.setup_plot()`: these methods are NOOP and can be overridden in sub-classes or monkey-patched to customise the plot display. It's necessary for [... cut]
closed 2017-11-20 by cjermain