pymeasure/pymeasure #128: Instrument.setting does not have set_process

bilderbuchi, 2017-12-22:
`Instrument.control` has `get_process` and `set_process`. `Instrument.measurement` has `get_process`. Why does `Instrument.setting` not have `set_process`? 

I think it would be useful, to e.g. cast truthy/falsy values to `1`or `0`, respectively. I know that I can (mis)use `validator` by passing an appropriate lambda, but it feels kludgy.

cjermain, 2018-01-02:
@bilderbuchi, the idea with these different function generators is that they each serve a specific case.

`Instrument.control` - a value that is set and read
`Instrument.measurement` - only read
`Instrument.setting` - only set

Does that clarify the difference?

Can you elaborate on the casting of boolean values? Where in the code are you suggesting?

bilderbuchi, 2018-01-02:
I know the difference, but found the absence of `set_process` inconsistent. To expand your list:

`Instrument.control` - a value that is set and read -> has `set_process` and `get_process`
`Instrument.measurement` - only read -> has `get_process`
`Instrument.setting` - only set -> does NOT have `set_process` - why?

My example: I have a power supply whose output is switched on/off by sending `OUT 1` and `OUT 0` (don't know exact command name right now). there is no output status query capability, so `Instrument.setting` is the appropriate function generator to use. Now, since this is in effect a boolean switch, and to keep a pythonic interface, I would like it to be able to supply "truthy" and "falsy" values, i.e. things that evaluate to `True` and `False`, respectively, to the property. All truthy values should become `1`, and all falsy values `0` in the command sent. 

Unfortunately, if I supply, e.g. `2` (which evaluates to boolean True) to my function, this gets turned into `OUT 2` (as it just passes the integer along), which is invalid. Therefore, I wanted to use `set_process` of `Instrument.setting`, which I expected to exist, to turn the supplied value into one that is valid for the device. If `set_process` existed, this would be, to my humble understanding from the other function generators and from the examples, the place to do such adaptation.

As a workaround, I have (mis)used `validator` and a lambda which basically casts all supplied values to bool, then to int - `lambda x, y: int(bool(x))` off the top of my hat (but don't have the code with me right now).

cjermain, 2018-01-04:
@bilderbuchi, thanks for the further explanation. I agree with your suggestion to add `set_process` to the arguments of `Instrument.setting`. Its absence was an oversight when I introduced the `get_process` and `set_process` to the other 2 functions. Do you want to start a PR?

bilderbuchi, 2018-01-04:
It will take a couple of weeks until I can start on this, but yeah, I'll try to get a PR together. :-)

PR #131 (2018-03-19): Add set_process to Instrument.setting. 
This fixes #128. Finally gotten around to do this.
commit 28c87bcf (2018-03-19): Add set_process to Instrument.setting.  (#131)

* Update .gitignore.

* Add set_process to Instrument.setting. Fixes #128.

* Bump required setuptools versions to avoid an error during testing.

See https://github.com/pytest-dev/pytest-runner/issues/39

* Add conda-forge channel to py3.4 requirements file.

Defaults does not have setuptools>27.3.0 for py34.
closed 2018-03-19 by cjermain