pymeasure/pymeasure #130: Version comparaison error in pyvisa

Bruyant, 2018-03-09:
Hi,

when using pyvisa I get and error on line 48 of pymeasure/adapters/visa.py.

It's a problem with visa version of the form '1.2.3' which cannot be converted to float.

I propose the following fix:

```
32a33,34
> from distutils.version import LooseVersion
> 
42c44
<     :param resource: VISA resource name that identifies the address
---
>     :param resource: VISA aresource name that identifies the address
48,49c50
<         version = float(self.version)
<         if version < 1.7:
---
>         if LooseVersion(self.version) < LooseVersion("1.7"):
155a157
> 
```

cjermain, 2018-03-15:
@Bruyant, thanks for your suggestion! I agree with your solution. Do you want to submit a pull request with this fix?

Bruyant, 2018-03-15:
I will try to learn how to do it.

bilderbuchi, 2018-03-15:
If you fork pymeasure to your github account, you can just edit the relevant file, save (it makes a branch and creates a commit automatically), and then follow the suggestions to create a PR (there is a button either directly after or when you display the branch that github just created for you)

cjermain, 2018-04-14:
@Bruyant, please update PyMeasure to version 0.5.1 and let me know if you still have this issue.

LastStarDust, 2018-09-19:
Hello. I am experiencing a similar issue:

 - **OS**: Centos 7 (Linux DAQ 3.10.0-862.11.6.el7.x86_64 #1 SMP Tue Aug 14 21:49:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux)
 - **pyvisa-py**: 0.3.1
 - **pyvisa**: 1.9.1
 - **pymeasure**: 0.5.1
 - **GPIB**: Bus 001 Device 002: ID 3923:709b National Instruments Corp. GPIB-USB-HS
 - **Device**: KEITHLEY INSTRUMENTS INC.,MODEL 2400
 - **linux-gpib**: 4.1.0-1.20180529svn1753.el7

**`pyvisa` + `pyvisa-py` seem to work fine (after much tinkering) but I cannot get pymeasure to recognize my instrument.** If there is additional info that I could report please let me know.

```
>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
(u'ASRL/dev/ttyS0::INSTR', u'GPIB0::24::INSTR')
>>> my_instrument = rm.open_resource('GPIB0::24::INSTR')
>>> print(my_instrument.query('*IDN?'))
KEITHLEY INSTRUMENTS INC.,MODEL 2400,4302945,C33   Mar 31 2015 09:32:39/A02  /V/M

>>> from pymeasure.instruments.keithley import Keithley2400
>>> sm=Keithley2400("GPIB0::24")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pymeasure/instruments/keithley/keithley2400.py", line 255, in __init__
    adapter, "Keithley 2400 SourceMeter", **kwargs
  File "/usr/lib/python2.7/site-packages/pymeasure/instruments/instrument.py", line 52, in __init__
    adapter = VISAAdapter(adapter, **kwargs)
  File "/usr/lib/python2.7/site-packages/pymeasure/adapters/visa.py", line 49, in __init__
    raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.
>>> pymeasure.__version__
'0.5.1'
```
PS Is there a way to use the GPIB device without sudo?

HikariBoy, 2019-06-19:
Let me begin with saying this is my first every post so I'm a noob here but I hope I can help..I had almost the identical problem as @LastStarDust except I could not connect to the USB port of an Agilent 33220A function generator.  Usng the following snippet I could connect and it correctly responded  
<code> import usbtmc</code>
<code> import subprocess</code>
<code> usb_devices = subprocess.check_output('lsusb').decode('UTF-8')</code>
<code> q = usb_devices.find('Agilent')</code>
<code> idVendor = usb_devices[q-10:q-6]</code>
<code> idProduct = usb_devices[q-5:q-1]</code>
<code> Agilent = usbtmc.Instrument(int(idVendor,base=16),int(idProduct,base=16))</code>
<code> print(Agilent.ask("*IDN?"))  </code>
RESULT: <b> Agilent Technologies,33220A,MY44035346,2.02-2.02-22-2 </b>  
However when I tried the following using pymeasure it failed:  
<code>import pymeasure  </code>
<code> from pymeasure.instruments.agilent import Agilent33220A  </code>
<code> sourcemeter= Agilent33220A('USB0::2391::1031::MY44035346::INSTR') </code>
RESULT:  The error 

> Please upgrade PyVISA to version 1.8 or later

I was using python3 so checked the distrubtion packages using <code> pip3 freeze </code> which showed:  
<b> PyMeasure</b>==0.6.1  
<b>pyserial</b>==3.2.1  
<b>pyusb</b>==1.0.2  
<b>PyVISA</b>==1.9.1  
<b>PyVISA-py</b>==0.3.1  
However it also showed <b>visa==1.0.0</b>  was installed.  
I used <code> sudo pip3 uninstall visa </code> which removed the <b>visa==1.0.0</b>  package and it now connects okay.  (Note: <b> serial==0.0.97 </b> was also installed which I mistakenly thought was a serial driver [it isn't, pyserial is] so I removed that as well however I don't believe that was the problem).  I gather this was an earlier version of pyvisa that changed name and so both packages existed (with incompatibility).  Hope that solves your problem (solved mine).

PR #284 (2020-06-06): Rename deprecated visa import to pyvisa.
Fix #257. Fix #130
closed 2020-06-06 by bilderbuchi