pymeasure/pymeasure #134: Connecting to instrument

jshoup1, 2018-04-13:
I am new to pymeasure. I am having the following issue when trying to connect/call the instrument.
Any help would be wonderful. 
Thanks
[image]

bilderbuchi, 2018-04-13:
This check is pretty fragile (as you found out). Better to use, e.g. `pkg_resources.parse_version()` (https://setuptools.readthedocs.io/en/latest/pkg_resources.html#parsing-utilities) to compare versions. Depends on `setuptools`, but pymeasure is using this already, anyway.

cjermain, 2018-04-14:
@jshoup1, thank you for reporting this. This is the same bug as #130. I will push a fix and release a new version to resolve this issue.

cjermain, 2018-04-14:
@jshoup1, please update PyMeasure to version 0.5.1 and let me know if you still have this issue.

jshoup1, 2018-04-16:
Thank you for the quick response. I updated and it is not working.

cjermain, 2018-04-16:
@jshoup1, can you be specific about the error?

bilderbuchi, 2020-03-06:
Let's close this as a duplicate of #130, and continue the discussion there.

closed 2020-03-06 by bilderbuchi