pymeasure/pymeasure #136: Supporting alternative output formats

mivade, 2018-04-17:
CSV is useful as a human-readable, easily-transportable file format and serves as a good default. Other formats (HDF5, SQL, numpy's format when using `np.save`, etc) also have their advantages and may be more appropriate for certain use cases.

One way to do this is to create a new abstract base `Formatter` class from which various formatters can inherit. Then the `Results` class can take an optional keyword argument `formatter` which tells it what formatter to use. This approach requires a bit of rework, though, since currently `Results` makes a lot of assumptions about using a plaintext CSV format (these assumptions make interacting with e.g. a SQL database more difficult).

Does anyone have any thoughts on this? Are there alternative approaches to supporting different output formats to consider?

mivade, 2018-04-17:
This would be good as a way of including many supported formats by default, but I think refactoring things a little to allow for custom solutions would be helpful. These ideas are not mutually exclusive, of course.

cjermain, 2018-04-19:
@mivade, great suggestion! I agree that CSV files are not ideal for every case. The main advantage is that experiment parameters can easily be stored alongside the data in the same file -- this is used extensively though PyMeasure so that old experiments can be opened and re-run. I would consider it a requirement for other storage formats to support storing the experiment parameters. HDF5 and SQL can certainly do this as well, but it will require some work to have consistent behavior.

tinix84, 2020-03-10:
im saving everything in mat file with this package https://pypi.org/project/mat4py/, for simple data is ok, if you need complex numbers or struct in matlab is better the scipy

carez, 2021-12-09:
Making pymeasure handle arbitrary data structures would be a huge improvement.

Can somebody with experience in the pymeasure codebase list the required changes to migrate to pandas/xarray, so we can maybe make a plan for a new Results/Formatter beyond csv?

bilderbuchi, 2021-12-10:
Ping @CasperSchippers, also see #277. 
I'd be careful to ask for "arbitrary datastructures", to be honest. Aside from that, a Pandas formatter/center point (i.e., put stuff into a DataFrame) that then can use Pandas' various export capabilities to output whatever formats Pandas supports (not only csv) sounds best/easiest, also since we already have pandas as a dependency.

CasperSchippers, 2021-12-11:
I agree that a Pandas center point might be nice (just to see if I understand you correctly, that would mean that the records would be cast into a DataFrame and handled from there; i.e. basically replacing the 'old-style'-formatter with a Pandas-based formatter like the one in PR #277?) to use it's export capabilities.

In general I don't see any issues with replacing the old-style `CSVFormatter` by the `CSVFormatter_Pandas` that PR #277 implements (while in the PR it is not a replacement, but rather an additional one; however, I don't see at this moment the use of the old formatter if we implement the new pandas-based formatter in such a way that it can handle both DataFrames and dicts as provided records, or maybe better convert from dict to DataFrame before handing it to the formatter).

For using other export capabilities I have two points of attention:
- At this moment, the way we actually perform the writing (i.e. using a FileHandler from the logging package) is only capable of handling plaintext files (at least, as far as I understood, but I might be mistaken of course); if we want to use the other export capabilities of pandas, that might require to get rid of the FileHandler and let pandas write the actual file (however we should think then of how to store the Parameters ((and e.g. the Metadata of PR #503)) in those files)
- At this moment, we rely on the fact that data can be appended in the datafile (i.e. we only write the new bits of data, without re-writing the old data); this works fine for CSV files (and probably also for a number of other plaintext filetypes), and it seems that this is also possible for some other types of formats (the Pandas exporter for e.g. HDF5 and SQL seems capable of doing so, but e.g. XML and excel not); if we also want to opt for these non-appendable filetypes, we need to either re-write the whole file again, every time, or write only after the measurement is finished. In either approach, we need to store the measured data in the DataFrame, which is not something we do right now
- Finally we also need to update the reading of the data (which is also very important for the live plotting, as we, at this moment, plot the data from the file, rather than directly from the emitted measurement-data)

Regarding the required changes to migrate to pandas; that fully depends on how far we want to migrate to pandas (or xarray for that matter, though I agree with @bilderbuchi that pandas is easier seeing what we have now): whether we want to let the actual writing of the data to a file be handled by the FileHandler as we do now (then I think it only requires a new formatter for another filetype, in which case I would be in favour of making an abstract Formatter class to base all other formatters on), or by Pandas for other, non-appendable non-plaintext types of file (in which case maybe the whole way we write our data needs revising).

mivade, 2021-12-11:
I don't really have a use case for this package anymore (at the moment, at least), but if you're already leveraging `logging`'s handlers I would think you could extend that a little bit and expose an interface that folks could use to define their own ways of writing and reading. As a convenience to users you could provide a handful of these handlers that can be selected that likely would leverage pandas internally.

> if we want to use the other export capabilities of pandas, that might require to get rid of the FileHandler and let pandas write the actual file

Related to my suggestion above you wouldn't use a `FileHandler` but instead a custom log handler, for example something like this:

```python
class CSVHandler(logging.FileHandler):
    def emit(self, record):
        df = pd.DataFrame(...)  # however you convert a log record into a DataFrame

        with open(self.baseFilename, self.mode) as buffer:
            df.to_csv(buffer)
```

and somewhere the user would specify an alternative handler if they want SQL, JSON, etc. What's nicer about this approach than just relying entirely on pandas is that if someone comes along and wants some format that's not supported by pandas for some reason (however unlikely) it's a pretty low barrier to entry.

BenediktBurger, 2024-06-19:
Pymodaq extracted their hdf5 storage facility into its own project, such that we could use that as well: https://github.com/PyMoDAQ/pydatamanager

SengerM, 2024-08-14:
Hello, I am considering to start using PyMeasure. Since I don't have experience with it, I would appreciate if some experienced user or developer can let me know whether it is currently possible to save data in a format other than CSV. 

My understanding is that the framework is designed in such a way that the data is automagically stored by the [`Results` class](https://pymeasure.readthedocs.io/en/latest/api/experiment/results.html#pymeasure.experiment.results.Results) but only in CSV. Should I write my code with PyMeasure and in the future I need a [`Procedure`](https://pymeasure.readthedocs.io/en/latest/api/experiment/procedure.html#pymeasure.experiment.procedure.Procedure) requiring other format than CSV, what would be the best way to proceed? One option would be to just save the data inside the `execute` method, and maybe only `emit` a fraction of the data for the graphical interface.

BenediktBurger, 2024-08-16:
The most simple way is, as you say, to save it in the Procedure itself.

All data is emitted via ZMQ as well, therefore, you could collect the data emitted via ZMQ and handle this data however you like.

Alternatively, you could modify the Results object itself.