pymeasure/pymeasure #137: Passing parameters to get_command and set_command in Instrument.control

vik-s, 2018-04-21:
I am writing an instrument driver for a source measure unit that has multiple channels. I want to use the Instrument.control method to do this because I really like the validators and map_values() capabilities. However, I need to change the get_command and set_command strings based on the channel I am writing or reading voltage or current.

I tried the following code, but it does not access the property / setter methods, but instead only returns a property object when I try to do `smu1_mode = channel_mode('SMU1')`.

Any help is appreciated!

` def channel_mode(ch):
        return Instrument.control(
            send_to_channel(ch) + "MODE?", send_to_channel(ch) + ":MODE %s",
            """ A string property that controls the channel mode,
            which can take the values 'V', 'I' or 'COMM'.
            VPULSE AND IPULSE are not yet supported""",
            validator=strict_discrete_set,
            values=["V", "I", "COMM"],
            check_set_errors=True,
            check_get_errors=True
        )`

cjermain, 2018-04-21:
@vik-s, great question! I have yet to test this scenario out. In principle, I would have expected your solution to work. There are two reasons why I suspect it might have failed:
1. Did you define `channel_mode` to be a `@staticmethod`?
2. Did you define `smu1_mode` as a class variable?

In order for the property to work properly it needs to be executed at the class level, before the object is constructed.

vik-s, 2018-04-23:
@cjermain, thanks for your input. I gave it some thought, and I think the correct way to do this is to put all the methods to control the channels in a separate class called `SmuChannel(Instrument)` which takes the channel name as an argument in its `__init__` method.

In the main `Agilent4156(Instrument)` class, I will create `SmuChannel` objects for each channel like:
```
self.smu1 = SmuChannel("SMU1")
self.smu2 = SmuChannel("SMU2")
``` 
and so on. This way, I will be able to call get/set methods defined in the `SmuChannel` class and it should access the attribute instead of returning a property object. The property calls would look like:
```
Agilent_Instrument.smu1.channel_mode
Agilent_Instrument.smu1.channel_mode = "V"
```
I'm hoping that this would work nicely, without having to repeat a lot of code for each channel in the main `Agilent4156` class. I will submit a PR once I get it working (which may take some time since I can only work on this on and off in my spare time), and report back here.

cjermain, 2018-04-23:
That sounds like a good approach. I'll keep an eye out for the PR.

vik-s, 2018-07-27:
The approach I described earlier worked well. Closing issue.

closed 2018-07-27 by vik-s