pymeasure/pymeasure #138: query about accessing time and using it as a parameter for continuous monitoring of a measurement system

ArtfulRefraction, 2018-04-23:
I am very much interested in using pymeasure for experiments in my lab. After having a look at the documentation it seems that  pymeasure is intended for experiments with a fixed number of iterations in mind. 

However, I would like to be able to measure/monitor certain parameters (e.g. temperature) as a function of time continuously for an undetermined amount of time. Would there be an easy way to implement this in pymeasure? 

Thank you very much in advance.

cjermain, 2018-04-23:
@ArtfulRefraction, that is a great question! You can set up an experiment to run without basing it on a fixed iteration.

If we take the basic script example from the repo, all you need to do is change the `for` loop to a `while` loop over some condition.
https://github.com/ralph-group/pymeasure/blob/13c557237d46b84197d2290efdabe51769ead309/examples/Basic/script.py#L39-L50

You could use `while(True)`, and then rely on the abort functionality (in `self.should_stop`) to stop the program. That would let it run forever. Alternatively, you can pass the while a function and determine it based on some other criteria.

The only side effect is the progress bar, which you manually update when calling `self.emit('progress', ...)`. You could use the modulus (`i % 100`) to just have it cycle through 100 steps to indicate that the program is running. We can certainly discuss a better UI strategy for this as well.

Do those suggestions solve the issue?

ArtfulRefraction, 2018-04-26:
@cjermain, thank you very much for your very prompt reply. It's great to know this project is very much alive!

I think what you propose should do the trick. In the meantime I have started adding a few instruments not currently in the repository and will be able to test what you propose once I have all the data acquisition electronics sorted. I will get back to you once I have done so.

Thank you very much once again!

cjermain, 2018-05-04:
@ArtfulRefraction, glad to hear it!