pymeasure/pymeasure #139: Line Feed for command

digrx, 2018-04-24:
I'm using agilent Spectrum Analyzer E4402B,
It seem's pymeasure not working with my SA because the command sent to SA not ended with line feed (0x0A character).
I'm using serial terminal to send *IDN? command and ended with 0x0A,
the SA reply with the identification.

cjermain, 2018-04-24:
Have you tried connecting to the instrument directly with the `VISAAdapter`?

digrx, 2018-04-27:
I was try using pyVISA directly and it's work using @py backend with command 
rm = visa.ResourceManager('@py')
I will try using VISAAdapter but I need specify to using @py backend.
How to do that from pymeasure ?

cjermain, 2018-04-30:
@digrx, the option to use the Python backend is not currently available. It would be easy to include by adding an argument to the `VISAAdapter.__init__` method, so that you can toggle between the VISA and Python backends. The `ResourceManager` is constructed here:
https://github.com/ralph-group/pymeasure/blob/13c557237d46b84197d2290efdabe51769ead309/pymeasure/adapters/visa.py#L55
I would suggest downloading the source for PyMeasure, making the change, and then `pip` installing it in editable mode (`pip install -e path/to/pymeasure`). That way you can try modifying the package and see the result.

bilderbuchi, 2022-04-06:
Yes, thank you. You would use the `visa_library` argument to define the desired backend ([docs](https://pymeasure.readthedocs.io/en/latest/api/adapters.html#pymeasure.adapters.VISAAdapter)).

closed 2022-04-06 by bilderbuchi