pymeasure/pymeasure #141: Problem with "open button" in the gui.py example

tvervust, 2018-05-04:
I am new to the pymeasure package and I just tried the gui.py example. This runs fine (using Spyder) but when I hit the open button I get the following error:

Traceback (most recent call last):
  File ".\Anaconda3\lib\site-packages\pymeasure\display\windows.py", line 325, in open_experiment
    dialog = ResultsDialog(self.procedure_class.DATA_COLUMNS, self.x_axis, self.y_axis)

  File ".\Anaconda3\lib\site-packages\pymeasure\display\widgets.py", line 364, in __init__
    self._setup_ui()

  File ".\Anaconda3\lib\site-packages\pymeasure\display\widgets.py", line 387, in _setup_ui
    self.layout().addWidget(preview_tab, 0, 5, 4, 1)

AttributeError: 'NoneType' object has no attribute 'addWidget'

My current installation setup:
- Anaconda version: 5.1.0; build py36_2
- Pymeasure version: 0.5.1

Any idea what went wrong?

cjermain, 2018-05-04:
@tvervust, what version of Qt do you have? It appears that the `layout` method is missing in `QtGui.QFileDialog` for your version of Qt.

tvervust, 2018-05-08:
Hello Colin,

These are the versions:
pyqt: 5.6.0
qt: 5.6.2

Is qt5 supported? I will try with qt4...

Thanks,
Thomas

cjermain, 2018-05-09:
@tvervust, it looks like in PyQt5, the `QFileDialog` had it's methods renamed, so I suspect the issue is related. See the discussion in http://pyqt.sourceforge.net/Docs/PyQt5/pyqt4_differences.html#qfiledialog.

The simplest short term fix is to use Qt4 and PyQt4.

It would be great to support Qt5 as well. I certainly welcome pull requests, and will see if I can find time to work on this feature.

bilderbuchi, 2020-03-06:
This should be solved with pymeasure 0.6

closed 2020-03-06 by bilderbuchi