pymeasure/pymeasure #143: Instrument programming model

vatgh, 2018-05-28:
Before contributing with some drivers for instruments I wanted to clarify the take of this project on which model to use. 
I could not find any reference to IVI or SCPI as a reference model in the documentation. Therefore, if I implement for example an oscilloscope, which model should I use in order to make the drivers for this oscilloscope work the same way as for all other oscilloscopes?
IVI addresses exactly this question (see http://www.ivifoundation.org/downloads/IVI_GSG_v_1.0.pdf in the introduction) with different classes for common instruments (oscilloscopes, multimeters, ...)
SCPI has only very basic common, defined comments, as far as I can tell (http://www.ivifoundation.org/docs/scpi-99.pdf)
Alternatively, one could use the structure that the commands of the instruments provide. But I think there is not necessarily large overlap from one instrument to the other. Small overlap would result in the need to modify code when swapping instruments which would be annoying.

I would prefer IVI as it addresses exactly this question and I do not see a need to reinvent the wheel. I would not implement all IVI functions for each instrument but only some basic ones to start with.

Btw I know that there is Python IVI (https://github.com/python-ivi/python-ivi) which has already a lot of driver implemented. Maybe one could make use of this.

bilderbuchi, 2018-05-28:
You might also take a look at InstrumentKit, which has [abstract instruments](https://github.com/Galvant/InstrumentKit/tree/master/instruments/abstract_instruments) like you suggest.

cjermain, 2018-05-28:
@vatgh, thanks for your thoughts on IVI vs. SCPI. We have typically focused on SCPI commands.

To give you more context, the PyMeasure project mainly focuses on the experiment execution and graphical user interfaces -- to make experiments easy to run and manage. For convenience, we have a fairly mature set of instrument classes. However, as @bilderbuchi suggested there are other projects that would be a better fit for deeper contributions on the instrument models. There has been some attempt to consolidate (#53), since many people are re-inventing the wheel on this problem. I think you are definitely on the right track with your approach.

bilderbuchi, 2018-05-29:
At the end of that discussion there are some links to further (hopefully more structured) discussion in the labpy organisation. In one issue we are currently trying to catalog the python instrument driver packages out there - help is welcome!

Making IK drivers available in pymeasure would be cool (there is an issue about this in the IK issue tracker), but I'm not aware of any progress on that front.

bilderbuchi, 2020-03-06:
See #94 for the issue about compatibility with IK

bilderbuchi, 2021-02-16:
This issue has more of a discussion/usage question character, so I will move it to our new discussion forum, which is a better fit than the issue tracker.

closed 2021-02-16 by bilderbuchi