pymeasure/pymeasure #151: Setting time-constant DSP7265 not working

CasperSchippers, 2018-07-31:
Whenever I try to set the time-constant of the signal-recovery DSP7265, I end up setting the wrong time-constant: it will revert to the minimal time-constant (10e-6 s) while i want to set e.g. 200e-3 s.
Here is a MWE:
```
from pymeasure.instruments.signalrecovery import DSP7265

lockin = DSP7265("GPIB::12")
lockin.time_constant = 200e-3
```

However, I found that when i replace the property time_constant of the DSP7265 with the exact definition from the DSP7265 instrument class, it seems to be working perfectly.

```
from pymeasure.instruments.signalrecovery import DSP7265
from pymeasure.instruments import Instrument
from pymeasure.instruments.validators import truncated_discrete_set

DSP7265.time_constant = Instrument.control(
    "TC.", "TC %d",
    """ A floating point property that controls the time constant
        in seconds, which takes values from 10 microseconds to 50,000
        seconds. This property can be set. """,
    validator=truncated_discrete_set,
    values=[
        10.0e-6, 20.0e-6, 40.0e-6, 80.0e-6, 160.0e-6, 320.0e-6,
        640.0e-6, 5.0e-3, 10.0e-3, 20.0e-3, 50.0e-3, 100.0e-3,
        200.0e-3, 500.0e-3, 1.0, 2.0, 5.0, 10.0, 20.0, 50.0,
        100.0, 200.0, 500.0, 1.0e3, 2.0e3, 5.0e3, 10.0e3,
        20.0e3, 50.0e3
    ],
    map_values=True
)

lockin = DSP7265("GPIB::12")

lockin.time_constant = 200e-3
```
Is this a bug or something in my setup?

cjermain, 2018-08-03:
@CasperSchippers, thanks for reporting this. This is certainly not expected. Without overwriting `time_constant`, are you able to write other values (e.g. `20.0`)? Which version of Python and PyMeasure are you using?

CasperSchippers, 2018-08-04:
I am able to set different values for the time constant, but not as intended (for example trying to set 1s results in a time constant of 20e-6, 2 gives 40e-6, and 3 gives 320e-6, so it also doesn't follow the order of the values in the definition of the parameter). Other properties of the lock-in seem to be working, so I can set e.g. the voltage
I use python 3.5.2 with pymeasure 0.5.1

cjermain, 2018-08-13:
Interesting... Does the instrument respond correctly to the explicit GPIB commands given manually? Can you add a print statement to the [`write` method](https://github.com/ralph-group/pymeasure/blob/4114450cb93f5bf435d771337bdcb40d79b0e748/pymeasure/instruments/instrument.py#L94) to see what command is being sent? You can edit your `lib/python3.5/site-packages/pymeasure` files directly for a quick trial (but be careful, because this is not a good practice in general!).

CasperSchippers, 2018-08-17:
I think i have found the issue, which is not on the writing part but rather on the reading part of the property. (Although I am quite certain that the writing also didn't work correctly at first, but that seems to be solved... magically; perhaps by changing the lock-ins (we use multiple), but i cannot reproduce the writing issue anymore....)

The issue with the reading of the time-constant lies in the commands that are used: the writing uses "TC %d", where d is an integer that is determined by a mapping from an actual time constant. the writing however is done using "TC." where the dot implies that an actual time (in seconds) is communicated, in stead of an integer that is mapped to a time. So I think it is as simple as replacing the "TC." by "TC" (or remove the mapping, and use a dot in the writing command, but I like the mapping better, gives a shorter command that is communicated). The same holds for the sensitivity.

I put the fix for both the sensitivity and the time-constant in a pull-request.

cjermain, 2018-08-18:
Nice catch! I'll review #155, and look to put out a new release with these changes.

bilderbuchi, 2020-03-05:
closed by #155

closed 2020-03-05 by bilderbuchi