pymeasure/pymeasure #160: How to handle \r on SCIP commands

robmarkcole, 2018-08-30:
As identified in https://github.com/ralph-group/pymeasure/issues/158 my Newport power meters require that every command be terminated with `\r`. I create this issue to discuss the solution. 
One approach is to create an `adapter` sub-class where for every command I `command += '\r` but this is very hacky. Alternatives?

bilderbuchi, 2018-08-30:
I also remember running into this previously (don't remember which instrument, though). I circumevented by appending the respective characters manually to each command, which is quite inelegant. I also did the custom Adapter with overridden `write` function, but that's just not nice.

Maybe a good approach would be to expose `write_termination` and `read_termination` parameters across adapters? Ideally both of them, because some instruments can be quite exotic concerning line endings. I seem to remember one that expected strings with `\r\n`, but returns terminated with `\n` (can't remember which one, though.

The canonical way with pyserial [seems to be](https://stackoverflow.com/a/16961872/599884) with io.TextWrapper (but use BufferedRandom instead of BufferedRWPair!). InstrumentKit has the capability to change the terminator (see https://github.com/Galvant/InstrumentKit/pull/130).

robmarkcole, 2018-08-30:
@bilderbuchi thanks for your advice, indeed the danfysik instrument implements a [custom adapter](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/instruments/danfysik/adapters.py)  which I can fork as a quick solution. Longer term I like the proposal of implementing `terminator` across all adapters. If we adopt pymeasure in our lab then I would be willing to do this.

Re InstrumentKit, whats the difference to pymeasure?

bilderbuchi, 2018-08-30:
I think you can leave this issue open for now, as the "right" way to do it seems to be clear but not yet implemented. I'll let @cjermain decide how he wants to handle this.

Re IK, check out #53 (warning: long read)

cjermain, 2018-08-31:
@robmarkcole, this is an interesting problem. The [Danfysik adapter](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/instruments/danfysik/adapters.py#L30) is certainly a good example, which required some more sophisticated logic on the read side to handle error reporting.  If you need that level of sophistication, then I think a dedicated class makes sense. 

For simple line termination, I like @bilderbuchi's suggestion for adding read and write termination strings. We could add them to the Adapter base class, and make sure they are implemented in child classes.

bilderbuchi, 2019-03-07:
I just ran (again I think) into the issue that the SerialAdapter cannot have custom read/write termination characters, so I had to postfix those to all the commands and strip manually from the results. Some "universal" solution would be appreciated.

bilderbuchi, 2019-03-07:
To add insult to injury, this was a device that had \r\n and \n read/write termination characters for normal commands, but the command to enable/disable remote control is actually a non-printing ascii character *without* termination, so it was actually "good" :cough: to have to suffix all the other commands manually... :P

bilderbuchi, 2019-03-18:
@cjermain would you think the SerialAdapter also learning `write_termination`/`read_termination` like the VISA adapter recently?

cjermain, 2019-03-18:
@bilderbuchi, I'm certainly open to adding in more extensive control for both terminations. Thanks for driving this discussion. Are you up for a PR?

bilderbuchi, 2019-03-19:
yeah, but it will take some time/weeks to get around to it, way too busy currently.

da66en, 2019-04-19:
Just got affected by this, too.

bilderbuchi, 2020-06-06:
So, a workaround for this issue is to supply the `read_termination`  and `write_termination` kwargs when creating the Instrument, and supply the port defintion as a string, and you will automatically use the VISAAdapter that can deal with this cleanly.

If, as an instrument developer, you want to make sure that the correct terminations are used then set the termination kwargs as defaults in the `__init__` argument list and pass them through the `super()` call. 

This only falls down when people supply an `Adapter` instance, not a port string, probably that is the point that will need some logic to handle. One could also argue that people passing in ready adapters should take care of terminations as they see fit (as they are probably troubleshooting, no?).

bilderbuchi, 2021-12-22:
We now have working and more flexible solution in place to deal with connection configuration when using the VISAAdapter (i.e. the majority of use cases). Already existing instrument drivers have been updated where appropriate.
If you run into this problem, please try with a VISAAdapter, first:
Please see https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#defining-default-connection-settings for instructions on how to implement an instrument driver with custom connection settings, and https://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html#modifying-connection-settings for how to override those settings as a user.

I'll close this issue, feel free to reopen/restart the discussion should you need this and can't use VISAAdapter.
Note: [this comment](https://github.com/pymeasure/pymeasure/issues/160#issuecomment-417287511) contains a proposed implementation approach.

closed 2021-12-22 by bilderbuchi