pymeasure/pymeasure #161: Error Code 800: Illegal with Storage Active

achadha0111, 2018-09-05:
I was trying to run the example procedure script after changing `sourcemeter.set_buffer(averages)` to `config_buffer(averages, 0.001)` since the class method no longer seems to exist. However, on running the procedure, I get **Error Code 800: Illegal with Storage Active** whenever the `start_buffer`, `wait_for_buffer` and `reset_buffer` command is called. I tried disabling storage on Keithley, yet that leads to no data being recorded.

cjermain, 2018-09-06:
@achadha0111, thanks for reporting this issue! It seems like you are referring to the [Making a measurement docs](https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html). The `Keithley2400` class does indeed have `set_buffer`, which comes from the inhereted [`KeithleyBuffer` class](https://github.com/ralph-group/pymeasure/blob/fb8e6e18c6a93ad867b6638987a976e1061690f7/pymeasure/instruments/keithley/buffer.py#L37). This class allows our `Keithley2000` class to have a consistent buffer interface.

Can you reproduce the issue by trying to send the buffer commands through the adapter: `sourcemeter.adapter.write(...)`?

achadha0111, 2018-09-13:
Thank you for the reply. My apologies, I was only able to go into the lab today to check. I changed the order of the statements in which the `reset_buffer()` function is called and it seems to work now.

closed 2018-09-13 by achadha0111

PYmeas, 2022-04-27:
Dear both, i am struggling with the same issue, could you please elaborate on the troubleshoot you implemented. i want to use that loop to record the average IV within a temperature loop and i get this error over and over. 
```
        for i in range(self.data_points):
            self.ivsweep.source_current = currents[i]
            self.ivsweep.reset_buffer()
            sleep(0.1)
            self.ivsweep.start_buffer()
            sleep(1)
            self.ivsweep.wait_for_buffer()
            sleep(1)
            # Record the average voltage and standard deviation
            self.ivsweep.measure_voltage() 
            sleep(1)
            voltages[i] = self.ivsweep.buffer_data.mean()
            sleep(1)
            voltage_stds[i] = self.ivsweep.std_voltage
            # self.ivsweep.measure_resistance()
            # sleep(0.1)
            # self.resistances[i] = self.ivsweep.resistance
            # resistance = self.ivsweep.buffer_data.mean()	
            now = datetime.now()
            self.record_csv(self.filepath, now, self.field, self.chamber, self.temperature, currents, voltages, voltage_stds)
        self.ivsweep.disable_source()

i use a init method for settings as follows:
```
def init_keithley(self):

        #######settings
        self.points = 5  #number of measurements per step (for statistics)
        # Set the input parameters
        self.data_points = 10  # number of steps in the sweep
        self.max_current = 0.001
        self.min_current = -self.max_current
        self.mode = 2  # 2 or 4 wires
        
        print("Initialize Instrument(s)...")
        self.ivsweep = Keithley2400("GPIB::24")
        self.ivsweep.reset()
        # self.ivsweep.source_mode = 'current'
        self.ivsweep.auto_range_source()
        self.ivsweep.use_front_terminals()
        self.ivsweep.apply_current()
        self.ivsweep.measure_voltage()
        self.ivsweep.source_current = self.min_current  # Sets the source current to min current mA
        # self.ivsweep.measure_resistance()
        # self.ivsweep.wires = 4
        # self.ivsweep.wires = 2
        self.ivsweep.compliance_voltage = 2.1
        self.ivsweep.source_current_range = self.max_current
        self.ivsweep.config_buffer(points = self.points, delay=0.1)
        self.ivsweep.reset_buffer()
        self.ivsweep.enable_source()

achadha0111, 2022-04-28:
It's been a while since I did this so my memory might be hazy but looking at your code, maybe you could try calling `reset_buffer()` in your `for` loop after you start the buffer.

If that doesn't work, could you give more details on where exactly the error happens?

PYmeas, 2022-04-29:
@achadha0111 thanks for the feedback it seems to solve the issue when using the code as below, no need to add a config_buffer in the init then. However the Reset() function of the pyvisa seems problematic and still generates error on query, so i removed the reset_buffer() and replaced it directly by the instruction self.ivsweep.write(":*CLS") in the init to clear the buffer, this way no more error :
````
for i in range(self.data_points):
      self.ivsweep.config_buffer(points = self.points, delay=0.1)
      self.ivsweep.source_current = currents[i]
      sleep(0.1)
      self.ivsweep.start_buffer()
      self.ivsweep.wait_for_buffer()