pymeasure/pymeasure #162: Connect over Ethernet bridge E5810x ?

MadaniJo, 2018-09-11:
Dear,
Is there an easy way/syntax to connect over this king of Ethernet/GPIB bridge like Agilent E5810A.
With python IVI syntax is like :
ivi.instrument('TCPIP::192.168.120.1::gpib0,2::INSTR') 
So I tried Keithley2400('TCPIP::192.168.1.159::gpib0,2::INSTR') but this fails
thanks.
Dan

physicsninja, 2018-09-14:
Hi Dan,

Could you post the error messages? I've never used that bridge before, but it definitely appears that that should be the correct command. If it works with that syntax when you instance a python IVI class, it should work the same for our adapter as we use pyVISA and that looks like the correct syntax.

MadaniJo, 2018-09-17:
Hi,
Have a look.
    def test_pymeasure(self):
        pwrmtr = Keithley2400("TCPIP::192.168.1.159::gpib0,2::INSTR")
        print(pwrmtr)

    def test_vxi11(self): 

        ip_address = '192.168.1.159'
        print(ip_address)    
        instr =  vxi11.Instrument(ip_address, "gpib0,2") 
        print(instr.ask("*IDN?"))

Second test work first not. Below the log. Thanks to for your interest.

MadaniJo, 2018-09-17:
The log:
E192.168.1.159
HEWLETT-PACKARD,E4418B,GB42421350,A1.08.01
.
======================================================================
ERROR: test_pymeasure (__main__.TestInstr)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "D:\test_git_repo\python-ivi\ivi\test\test_ivi_dizic01.py", line 25, in test_pymeasure
    pwrmtr = Keithley2400("TCPIP::192.168.1.159::gpib0,2::INSTR")
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pymeasure\instruments\keithley\keithley2400.py", line 255, in __init__
    adapter, "Keithley 2400 SourceMeter", **kwargs
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pymeasure\instruments\instrument.py", line 52, in __init__
    adapter = VISAAdapter(adapter, **kwargs)
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pymeasure\adapters\visa.py", line 65, in __init__
    **kwargs
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Py36-32_VirtEnv\VirtEnv_02\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

----------------------------------------------------------------------
Ran 2 tests in 0.171s

FAILED (errors=1)

MadaniJo, 2018-09-21:
Looks nobody as a solution for this ?

physicsninja, 2018-09-21:
Sorry, for not getting back to you, it was a hectic week. 

I don't know what the root cause of the problem is. I'm not sure it is the optimal solution, but perhaps the simplest solution is writing an IVIadapter using the VISAadapter as a template, given that you know the call to the IVI library works. Given that the IVI library is modeled to fall back to VISA, I imagine things would be pretty similar. Then you could either modify the Keithley 2400 to have an optional argument which uses the IVIadapter instead of the VISA adapter, or just change it locally on you computer. 

Any thoughts, @cjermain ?

cjermain, 2018-09-24:
@MadaniJo, the Keithely2400 uses VISA by default, when the adapter is not specified. That means that your address needs to work when using VISA alone. I would suggest using an IPython (Jupyter) notebook or terminal to try out the commands with VISA.

If you need IVI specifically, @physicsninja's suggestion of writing a new adapter is the best option. [Adapters](https://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html#using-adapters) encapsulate the communication to an instrument, so that instrument commands can be independent of the communication method. I'm happy to review a PR for a IVIAdapter class. Check out [VISAAdapter](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/adapters/visa.py#L39) as a starting point, and feel free to reach out with any question.

thosou, 2018-10-18:
Hi,

I ran into the same issue. I found out that when using a VisaAdapter with the NI library it does not work while using the py backend it does. There are apparently few things to set up in order to get that to work with the NI lib (https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019LccSAE).
Unfortunately I use linux for my measurements so I did not try it out.
I have also noticed that the behavior is better when using vxi library than the py backend. So my colleague and I made an adapter and it works fine. I believe that if no one has started something yet I can share that.
Let me know.
/Thomas

MadaniJo, 2018-10-18:
Hi,
I would have a look and test your adapter if you push it.
Regards,
Dan

cjermain, 2018-10-22:
@thosou, thanks for your input. I'm happy to review a PR for your VXI adapter, and it sounds like @MadaniJo would also be able to review.

bilderbuchi, 2020-03-05:
should be closed by #166

closed 2020-03-05 by bilderbuchi