pymeasure/pymeasure #167: Streaming data in Jupyter using streamz

robmarkcole, 2018-10-26:
HI 
thought I would highlight this approach to streaming data in a Jupyter notebook, which allows considerable scope for experimentation with the plot style

http://pyviz.org/tutorial/11_Streaming_Data.html

[image]

cjermain, 2018-11-28:
@robmarkcole, this is a cool package. Thanks for bringing it up.

bilderbuchi, 2020-03-05:
@robmarkcole this looks interesting. I think this could be part of the documentation as a [tutorial](https://pymeasure.readthedocs.io/en/latest/tutorial/index.html) - could you add/make a [PR](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html#making-a-pull-request) for that?

robmarkcole, 2020-03-05:
Hi @bilderbuchi  my wife is expecting a baby any day now so I cant make any promises

bilderbuchi, 2020-03-05:
No worries, no promises were expected. Best of luck!