pymeasure/pymeasure #168: ZMQ error

jmittelstaedt, 2018-11-02:
Hi,

While running PyMeasure on a newer version of anaconda, I got an error which looks like it's coming from an update ZMQ had. In particular, the error occurs when a `worker` tries to use its `emit` method. The error is in the `self.publisher.send_serialized` function, which I'm guessing has changed in a newer version of ZMQ or something. A fix for now is to add a `TypeError` to the following `except` statement, but some functionality (which it seems I don't use) is probably being lost along the way. The full error message is at the bottom of this post, and the versions of everything that I'm using are below.

ZMQ version: 17.1.2
Python version: 3.6.6
Conda version: 4.5.11

`Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\zmq\sugar\socket.py", line 427, in send_multipart
    memoryview(msg)
TypeError: memoryview: a bytes-like object is required, not 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Joseph\Anaconda3\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\pymeasure-0.5.1-py3.6.egg\pymeasure\experiment\workers.py", line 161, in run
    self.update_status(Procedure.RUNNING)
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\pymeasure-0.5.1-py3.6.egg\pymeasure\experiment\workers.py", line 115, in update_status
    self.emit('status', status)
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\pymeasure-0.5.1-py3.6.egg\pymeasure\experiment\workers.py", line 95, in emit
    self.publisher.send_serialized((topic, record), serialize=cloudpickle.dumps)
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\zmq\sugar\socket.py", line 512, in send_serialized
    return self.send_multipart(frames, flags=flags, copy=copy, **kwargs)
  File "C:\Users\Joseph\Anaconda3\lib\site-packages\zmq\sugar\socket.py", line 434, in send_multipart
    i, rmsg,
TypeError: Frame 0 (128) does not support the buffer interface.`

lawa42, 2019-05-02:
Same problem here

zmq: 18.0.1
zeromq: 4.3.1
python: 3.7.3
pymeasure: 0.6.0

CasperSchippers, 2019-05-22:
I've got the same problem here as wel, but i figured out that the problem only arises when you have cloudpickle installed (for reference: in my case this is cloudpickle 1.0.0)

cjermain, 2019-06-04:
@jmittelstaedt and @lawa42, thanks for reporting this issue. It sounds like there is an incompatibility with `cloudpickle` as @CasperSchippers brings up. I will take a look when I get the opportunity.

cjermain, 2019-06-17:
This could also be an issue with Python 3.7, since we don't have any test visibility yet.

@jmittelstaedt, can you put together a minimal example that displays the behavior, and we can add it to the unit tests.

Kassikakk, 2019-06-19:
I've looked into this issue and it seems that for some reason the send_serialized function in zmq wants the serialization function result not as a single frame, but as a list (or other iterable) of those (perhaps it has changed some time if it worked earlier?), which cloudpickle.dumps itself doesn't provide. An obvious solution would be to change line 95 in workers.py from

`
	self.publisher.send_serialized((topic, record), serialize=cloudpickle.dumps)
`
to
`
	self.publisher.send_serialized((topic, record), serialize=lambda x : [cloudpickle.dumps(x)])
`

Corresponding adjustments are then also needed for any receiving party. Currently I think the only places within the package where recv_serialized is used is in the Listener/QListener classes. So using this as an example, in /experiment/listeners.py line 81, instead of

`
	topic, record = self.subscriber.recv_serialized(deserialize=cloudpickle.loads, flags=flags)
`
one should do
`
	topic, record = self.subscriber.recv_serialized(deserialize=lambda x : cloudpickle.loads(x[0]), flags=flags)
`

(and there is a similar definition of QListener class in /display/listeners.py). 

Also, if I understand correctly then any topic filtering can only be done after deserialization, i.e. the first call to setsockopt (zmq.SUBSCRIBE,'') has to be with an empty string.

There may of course be some side effects I'm not aware of, but generally this way the zmq pub/sub seems to be working.

very3b, 2020-05-23:
> I've got the same problem here as wel, but i figured out that the problem only arises when you have cloudpickle installed (for reference: in my case this is cloudpickle 1.0.0)
> 
> To solve this issue (for the time being) you could either remove the cloudpickle package from your setup or prevent importing the cloudpickle module (though might not be a very neat solution), e.g. using:
> 
> ```
> import sys
> sys.modules['cloudpickle'] = None
> ```
> 
> Both solutions seem to be working for me, but are of course not real long-term solutions.

works for me, thanks

bilderbuchi, 2020-10-20:
@CasperSchippers can you check if you still get this error? There have been a number of cloudpickle releases in the meantime, maybe this got fixed and we just need to bump the minimum required version?

CasperSchippers, 2020-10-20:
I just checked, but I do indeed get the error still with cloudpickle version 1.6.0 (i.e. latest version).

Kassikakk, 2020-10-20:
We probably shouldn't expect it to be fixed as this isn't a bug but a documented feature and indeed that of [zmq's send_serialized](https://pyzmq.readthedocs.io/en/latest/api/zmq.html#zmq.Socket.send_serialized) (and correspondingly recv_serialized):

serialize (callable) – The serialization function to use. serialize(msg) should return **an iterable** of sendable message frames (e.g. bytes objects), which will be passed to send_multipart.

As it stands, serialize (i.e. cloudpickle.dumps) only returns a frame, so it needs to be converted to an iterable, e.g. list. I actually use zmq communication and for me the solution I proposed earlier, works (also considering the remark about topic filtering). In fact I've been having to fix it every time I install or upgrade pymeasure somewhere. If anyone has a better idea, be my guest.

bilderbuchi, 2020-10-20:
OK then; is anybody up to implement a test showing this behaviour and proposing a fix?

bilderbuchi, 2020-10-27:
Some datapoints from #353:
* Can be triggered by running e.g. `gui.py`, `image_gui.py`
* It looks like when cloudpickle and pyzmq are both installed this error is thrown (both are brought in by Spyder).
* If just one or the other is installed the code runs error free.

bilderbuchi, 2020-10-27:
@Kassikakk could you elaborate on this point?
> Also, if I understand correctly then any topic filtering can only be done after deserialization, i.e. the first call to setsockopt (zmq.SUBSCRIBE,'') has to be with an empty string.

What goes into the topic arg of the constructor is most probably a string anyway, so why does it have to be empty first, and what does that have to do with deserialisation? Curiously, I could not find a spot where [Q]Listener are used, but I am not familiar with this part of pymeasure.

bilderbuchi, 2020-10-27:
OK, I started a fix in #354, but I am getting weird issues in the test suite. Maybe someone else can help, I am not really familiar with that part of the codebase. @CasperSchippers ?

Kassikakk, 2020-10-28:
In order for the built-in topic filtering to work, the topic as specified in `socket.setsockopt(zmq.SUBSCRIBE, topic)` must come at the beginning of the incoming dataframe, which however is at that point serialized, i.e. in the current implementation the result of `cloudpickle.dumps((topic,record))`. Now if you try to substitute a string for topic (and anything for record) and view the returned result of this function, you see that the topic string is there somewhere, but not at the beginning, meaning that it will not pass the topic filter. Hence one way to get the data is to pass an empty string to the filter (thus letting through all the messages) and later when you have the deserialized result (topic, record), check if the topic is of interest and correspondingly do something with (or skip) the record. A benefit of this method is that you can actually monitor multiple topics with one listener.

A possibly cleaner method that might work (haven't tested it myself) is suggested in this [SO article](https://stackoverflow.com/questions/24048319/setting-the-topic-when-using-the-pyobj-subfunctions-in-zeromq-python) (a bit of discussion is added [here](https://stackoverflow.com/questions/33159395/how-to-add-topic-filters-when-calling-recv-pyobj-in-zeromq)). Apparently, if a multipart message is received and the first part has (or perhaps begins with) the filter string, the message will pass through. So one can set the first part to be the topic and add the serialized record in a second part. If one still wants to use send_serialized/recv_serialized, it would be something like

```
#publisher (Worker) part
self.publisher.send_serialized(record, serialize= lambda x: [topic, cloudpickle.dumps(x)])

#subscriber (Listener) part
topic, record = self.subscriber.recv_serialized(deserialize=lambda x : [x[0], cloudpickle.loads(x[1])])

```
but as these functions call the send/recv_multipart anyway, it may make more sense to call them directly, as in the referred example (replace 'pickle' with 'cloudpickle' there).

CasperSchippers, 2020-10-31:
> OK, I started a fix in #354, but I am getting weird issues in the test suite. Maybe someone else can help, I am not really familiar with that part of the codebase. @CasperSchippers ?

I've had a look. On my pc, the new `test_zmq_serialisation` seems to run fine itself, but for some reason the test then hangs at another test. It is however not consistent at which test it precisely hangs, and all of the tests that I've seen it hanging at also work perfectly fine if the `test_zmq_serialisation` is removed.
At this moment, I couldn't find a way to solve this problem, but admittedly I'm also not very familiar with this part of the code.

bilderbuchi, 2020-11-01:
Yeah, by now I suspect that we aren't cleaning up ZMQ resources correctly at the end, which is why the test hangs. It also happens on CI, the builds time out. I'll have to investigate that.

bilderbuchi, 2021-03-01:
So, fun, I think I fixed all the problems (available in #249), and it works locally on Ubuntu, but the related CI test on Windows fails and I don't know why. Can anybody help? Is there some material difference in socket handling?

bilderbuchi, 2021-03-02:
All done now, please check out #354

PR #354 (2021-03-03): Fix ZMQ bug
Closes #168.

Let's first see that we can trigger the problem.
closed 2021-03-03 by bilderbuchi