pymeasure/pymeasure #172: Add Prologix gpib to ethernet adaptor

mbrad092, 2019-01-04:
Also the VXI-11 adapter isn't present in the pip version of this package. Thanks for your efforts, really enjoying the package!

thosou, 2019-01-05:
Hi @mbrad092,

I actually also use the ethernet to gpib prologix adapter! I made a personal quick and dirty solution based on what has been done with the serial prologix. I had in mind to open a discussion and see how we could add this contribution.
Regarding the VXI-11 adapter, I think it will be probably available in the next release. until then you can always install from github.

mbrad092, 2019-01-09:
Oh lol very nice! Any chance I could ask you for to send me your quick and dirty solution in the interim before the contribution is finalized?

thosou, 2019-01-10:
Hi @mbrad092,

You can find the code here:
https://github.com/thosou/pymeasure/blob/dev/prologix-ethernet/pymeasure/adapters/prologix.py

You can use this adapter the same way as the `PrologixAdapter`
Let me know if you get it to work!

cjermain, 2019-01-15:
@mbrad092, thanks for reminding me to make a new release! I've published version 0.6 which now has the VXI-11 adapter.

@thosou, it would be great to add your PrologixEthernetAdatper to PyMeasure.

bmoneke, 2023-02-01:
@msmttchr 
I compared the code @thosou mentioned to our current PrologixAdapter. It should be quite simple to add some good ideas (making auto, eoi, eos configurable and readable).
That way we make the Adapter capable to use the Ethernet version (which seems to use different eoi and eos setting, at least as default) as well.

msmttchr, 2023-02-01:
I think this is a good idea 😃

bmoneke, 2023-02-08:
Whoever wants to use the Prologix Adapter with ethernet: the code should work.
Please tell us, if not.

BenediktBurger, 2025-06-27:
It should work.
I'll close this issue.
Whenever someone has an issue with that one, he might raise a bug issue.