pymeasure/pymeasure #173: Writing different results to multiple files

nickagian, 2019-01-10:
Hi,

In my set-up I would like to write results to two different files, but not in the way it was described in #75. 

Let's say for example that in the "Current-Voltage measurement" example, except for the 'Current, Voltage, Resistance' set of measurements, I would like to take some current consumption measurements before and after collecting the above data. It doesn't necessarily make sense in this case, but it makes for my set-up.

So I would like to have this values to a different csv file.

Is that possible?

physicsninja, 2019-01-14:
Sorry for not replying to your issue sooner. 

You can definitely do it. I'm not sure what the most maintainable way to do it is (perhaps @cjermain is a better person to ask), but I can suggest the way that I did it. I use a custom gui like in this [example](https://github.com/ralph-group/pymeasure/blob/master/examples/Basic/gui_custom_inputs.py)

Just to restate my understanding of the problem, you are measuring (continuing with the above example) V and R as a function of applied I, for instance, and before and after a set of those measurements you want to measure of I vs V and save those results to a different file.

The way I have approached such a problem in the past is to have a boolean parameter in the procedure definition that turns on one measurement or the other. Using the example:

    max_current = FloatParameter('Maximum Current', units='mA', default=10)
    min_current = FloatParameter('Minimum Current', units='mA', default=-10)
    current_step = FloatParameter('Current Step', units='mA', default=0.1)
    delay = FloatParameter('Delay Time', units='ms', default=20)
    voltage_range = FloatParameter('Voltage Range', units='V', default=10)
    #the new parameter
    current_consumption_scan = BooleanParameter('do I consumption scan?', default = False)

    DATA_COLUMNS = ['Current (A)', 'Voltage (V)', 'Resistance (Ohm)']

    def startup(self):
        log.info("Setting up instruments")
        self.meter = Keithley2000("GPIB::25")
        self.meter.measure_voltage()
        self.meter.voltage_range = self.voltage_range
        self.meter.voltage_nplc = 1 # Integration constant to Medium
        
        self.source = Keithley2400("GPIB::1")
        self.source.apply_current()
        self.source.source_current_range = self.max_current*1e-3 # A
        self.source.complinance_voltage = self.voltage_range
        self.source.enable_source()
        sleep(2)

    def execute(self):
        currents_up = np.arange(self.min_current, self.max_current, self.current_step)
        currents_down = np.arange(self.max_current, self.min_current, -self.current_step)
        currents = np.concatenate((currents_up, currents_down)) # Include the reverse
        currents *= 1e-3 # to mA from A
        steps = len(currents)

        #create the other set of independent variables
        volts_up = np.arange(self.min_volts, self.max_volts, self.volts_step)
        volts_down = np.arange(self.max_volts, self.min_volts, -self.volts_step)
        volts = np.concatenate((volts_up, voltss_down)) # Include the reverse
        volt*= 1e-3 # to mV from V
        steps = len(volts)

        #add our conditional
        if not current_consumption_scan:
            ....... #do the original scan as in the example
        if current_consumption_scan:
             ...... 
             #set up our instruments to measure current consumption instead
             #do our new current consumption scan

    def shutdown(self):
        self.source.shutdown()
        log.info("Finished")

then in the queue method of the main window create a sandwich of scans

 
        #queue the current consumption scan
        filename = unique_filename(directory, prefix='I_consumption_before')
        procedure.max_volts = self.inputs.max_volts.value()
        ....
        procedure.current_consumption_scan = True
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

        #queue normal scan
        filename = unique_filename(directory, prefix='Normal_scan')
        procedure.max_current = self.inputs.max_current.value()
        ....
        procedure.current_consumption_scan = False
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

        #queue after current consumption scan
        filename = unique_filename(directory, prefix='I_consumption_after')
        procedure.max_volts = self.inputs.max_voltsvalue()
        ....
        procedure.current_consumption_scan = True
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

This will queue a current consumption scan, then the normal scan, then another current consumption scan and save each one with a different file prefix. I used this in the context of queueing up multiple similar scans but needing to do something special for the first one only (waiting for temperature equilibration) so I would set the equilibrate temperature boolean to true only for the first one in the set of procedures. Let me know if this works for you or if you have any further questions

bilderbuchi, 2021-02-16:
This issue has more of a discussion/usage question character, so I will move it to our new discussion forum, which is a better fit than the issue tracker.

closed 2021-02-16 by bilderbuchi