pymeasure/pymeasure #180: Faulty function get_buffer() in dsp7265

badarirao, 2019-02-14:


cjermain, 2019-02-20:
@badarirao, thanks for raising this issue! The `get_buffer` method indeed has some errors as implemented. The `self.length` is indeed not defined, but appears as if it should be a property that returns the length of the buffer. I would recommend that you get the DSP7265 manual and try to manually run the commands. Once we have the working set, we can revise this method.

badarirao, 2019-02-21:
@cjermain ,Thanks for looking into this bug.
Could you give me the algorithm on how the function is supposed to work? I am not exactly able to understand the given code.
Also, the signal recovery manual says something about implementation of serial poll in order to extract the data. I am not exactly sure on how to do this. The exact note in the manual is as follows:
" Note that when using this command with the GPIB interface the serial poll must be
used. After sending the DC command, perform repeated serial polls until bit 7 is set,
indicating that the instrument has an output waiting to be read. Then perform repeated
reads in a loop, waiting each time until bit 7 is set indicating that a new value is
available. The loop should continue until bit 1 is set, indicating that the transfer is
completed. "

Can you point me in the right direction on how to implement this?

Thanks,
Badari

cjermain, 2019-02-26:
Unfortunately, I did not write the original function, nor do I have access to that instrument.

I would suggest determining which command provides the status, and then do a bit comparison to find the value the manual suggests. You will need to use the `DC` command to pull back the values. All of this can be done by directly calling read/write/ask on the instrument, and manually stepping through the process.

bilderbuchi, 2020-10-25:
@badarirao did you every manage to fix the issues you report here?

badarirao, 2020-10-25:
@bilderbuchi  No I could not fix this issue.

bilderbuchi, 2021-03-02:
@CasperSchippers do I remember correctly that you are using such a device? If so, could you maybe take care of fixing this?

CasperSchippers, 2021-03-02:
I have used it in the past, and I think I can get access to this system; I'll have a look

CasperSchippers, 2021-03-22:
I started a PR (#418) to solve the issue

closed 2021-04-19 by CasperSchippers