pymeasure/pymeasure #182: Bug to set Keithley 2400 in measure resistance mode

ronan-sensome, 2019-03-07:
In the module: `pymeasure.instruments.keithley.keithley2400`, the method `measure_resistance` in class `Keithley2400`, single quotes are missing in the command sent line 276 that reads

```python
self.write(":SENS:FUNC RES;"  # missing single quotes
           ":SENS:RES:MODE MAN;"
           ":SENS:RES:NPLC %f;:FORM:ELEM RES;" % nplc)
```

It should be replaced by 

```python
self.write(":SENS:FUNC 'RES';"
           ":SENS:RES:MODE MAN;"
           ":SENS:RES:NPLC %f;:FORM:ELEM RES;" % nplc)
```

cjermain, 2019-03-10:
Thanks for raising this @ronan-sensome! Why are the quotes needed in this context?

ronan-sensome, 2019-03-10:
Hello
The :SENS:FUNC need a string as argument. I have tried to set my keithley 2400 to measure resistance using this function and I have got a data type error. After checking,  I have remarked that the function `measure_voltage` which is very similar to `measure_resistance` use the same command with the argument 'VOLT' (with quotes).
The datasheet confirms my idea, we need quotes to make the argument valid.
Here a link of a quick manual where you can find example where this function is called (p18 of this manual which is page 26 of the pdf). 
https://sites.fas.harvard.edu/~phys191r/Bench_Notes/A1/2400_quick.pdf

I hope it will help
Best regards

cjermain, 2019-03-10:
This has been resolved in #183.

closed 2019-03-10 by cjermain