pymeasure/pymeasure #186: Keithley buffer - is_buffer_full function does not work

hlgirard, 2019-03-14:
The `is_buffer_full()` function of the Keithley buffer module always timeouts.

Trying the run `sourcemeter.ask('*STB?')` returns `0\n` both during the buffer filing and when it is full.

My setup is pymeasure (v0.6.0) -> Prologix 6 GPIB controller -> Keithley2400.

bilderbuchi, 2020-06-06:
@hlgirard did you ever get to the bottom of this? Possibly you have to enable that the SMU even flags this, I find in the manuall of the 2400 the table 15-8 "Program and read register programming example", where the buffer-full bit is enabled:
```
Command                         Description
FORM:SREG BIN - Select binary format to read registers.
STAT:MEAS:ENAB 512 - Enable BFL (buffer full).
STAT:MEAS:COND? - Read Measurement Condition Register.
STAT:MEAS? - Read Measurement Event Register.
```
so possibly one has to enable that register first.

hlgirard, 2020-06-07:
@bilderbuchi I ended up working around this by implementing the buffer client-side since I didn't need the additional sampling frequency made possible by instrument-side buffering.

Your comments make a lot of sense, unfortunately I non longer have access to the instrument so I won't be able to test them.

bilderbuchi, 2020-06-10:
OK thanks. I have access to a device - do you know a quick way to fill up the buffer, so that I can test any fixes I make?

hlgirard, 2020-06-14:
It has been a while but if I remember correctly, I used to set the sampling program on the instrument itself (say 1000 points over 2 seconds) and trigger programmatically.

The idea was to use the very fast sampling speed available to the instrument itself by doing a cycle of triggering, waiting until the buffer is full, retrieving the data, erasing the buffer, and triggering again.

bilderbuchi, 2021-03-02:
I don't have acess to a test device anymore, so I don't know what to do about this anymore. The function itself uses a straightforward SCPI command, so I suspect that this was a problem of the device (configuration). Please feel free to re-open if you encounter this again.

closed 2021-03-02 by bilderbuchi