pymeasure/pymeasure #194: Newport ESP300 read write termination / VI_ERROR_TMO (-1073807339) Timeout #157

hududed, 2019-05-02:
Hi all

I am running this old ESP300 machine which is included in pymeasure instruments.

I seem to be running into a `VI_ERROR_TMO (-1073807339) Timeout #157` issue and the solutions seem to be keying in the right terminations. How do I achieve this with the ESP300 package?

I have tried `my_instr = ESP300("GPIB0::3::INSTR", read_termination = '\r\n', write_termination = '\r')` but I don't think it worked - I don't know how to check whether it worked.

Please advise.

Cheers, H

jmittelstaedt, 2019-05-04:
Hi,

I have been plagued by VISA timeout errors with the ESP300 for a while now. Unfortunately, I don't think the read/write terminations are the problem since I have been able to communicate with it (mostly) successfully without  specifying them explicitly. However, if you do want to check that they are set, I think you can do so by looking at the corresponding attributes of the VISA connection object directly, which you can access as follows:
```
my_instr = ESP300("GPIB0::3::INSTR", read_termination = '\r\n', write_termination = '\r')
my_instr.adapter.connection.read_termination
my_instr.adapter.connection.write_termination
```
and hopefully the last two statements will return the read and write terminations that VISA is using to communicate to your instrument.

The only solution I have found to deal with the timeouts is to do something like
```
esp_faliure_counter = 0
esp_success = False
FALIURE_LIMIT = 10
while esp_faliure_counter < FALIURE_LIMIT:
    try:
        # whatever communication you need to do with the ESP300
    except pv.errors.VisaIOError:
        esp_faliure_counter += 1
        log.warning(f"failed at count {esp_faliure_counter} of {FALIURE_LIMIT}")
        )
        continue
    esp_success = True # only runs if we beat try statement
    break
if not esp_success:
    raise RuntimeError("Not able to successfully communicate with ESP300")
```

The ESP300 has quickly become my least favorite instrument to communicate with, I hope you have better luck than I do.

bilderbuchi, 2020-03-05:
@hududed has your question been resolved?

bilderbuchi, 2020-06-06:
No response from the reporter, and I agree with @jmittelstaedt's assessment that this probably has nothing to do with termination, so closing this. Feel free to re-open if you still have this issue.

closed 2020-06-06 by bilderbuchi