pymeasure/pymeasure #214: Polytec MSA

Endever42, 2019-07-22:
Hi,

I am currently implementing a python based control class for the Polytec MSA (micro system analizer). This is tricky, since it uses some windows api functions to do this and the documentation is terrible. This also means, that you cannot connect to the device, but have to run Python locally on the machine (or in my case winpython from a USB stick). So, I am not sure if this fits the pymeasure project, but if it does, I would be happy to share this.

Endever42, 2019-08-05:
Any comments on this? Otherwise I am thinking of creating a project for this myself, but in combination with pymeasure, the exposure would be higher.

bilderbuchi, 2020-03-05:
@cjermain do you have an opinion about this?

Endever42, 2020-03-05:
I would also like to add, that I had some discussion with Polytec and they would support this (only in terms of logos and awareness).

bilderbuchi, 2021-02-16:
I'm not sure how well this fits within Pymeasure, but feel free to re-open if you disagree. :-)

closed 2021-02-16 by bilderbuchi