pymeasure/pymeasure #216: Example iv_keithley.py gives Error

JH0894, 2019-07-25:
Hey everyone,

I'm new to PyMeasure and I am trying around a bit with the examples to get a feel for how it works. I tried to run the iv_keithley.py example that I found here on GitHub with a Keithley that I have. I changed the GPIB address and changed the directory. When I run the file, the window shows up. However, when I click on "queue" it gives this error (see picture). Does anyone know how to solve this? Thanks in advance!

[image]

jmittelstaedt, 2019-09-01:
Hi,

I believe this is the same error as #168.

Some hotfixes were suggested in that thread.

bilderbuchi, 2020-03-05:
Yes, this is #168, please track that issue instead.

closed 2020-03-05 by bilderbuchi