pymeasure/pymeasure #221: error in pymeasure with keithley2450

20360, 2019-08-30:
I can run the code following this link;
https://pymeasure.readthedocs.io/en/latest/api/instruments/keithley/keithley2400.html

_from pymeasure.instruments.keithley import Keithley2450
keithley = Keithley2450("USB0::0x05E6::0x2450::04407378::INSTR")
keithley.apply_voltage()                # Sets up to source current
keithley.source_voltage_range = 10   # Sets the source current range to 10 mA
keithley.compliance_current = 1e-3         # Sets the compliance voltage to 10 V
keithley.source_voltage = 0          # Sets the source current to 0 mA
keithley.enable_source()                # Enables the source output
keithley.measure_current()              # Sets up to measure voltage
keithley.ramp_to_voltage(10,11,1)          # Ramps the current to 5 mA
print(keithley.voltage, keithley.current)                 # Prints the voltage in Volts_

But, code following link makes error ('Keithley2400' object has no attribute 'config_current_source')
https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html

It seems that the libraries in two link are different because I have no library about "config_current_source".
How can I download the library in second link?

jmittelstaedt, 2019-09-05:
Hi,

I think that the code in the second link you gave is old, and the `Keithley2400` object has had the `config_current_source` attribute removed or changed. The first link should have the documentation for the most recent version of the instrument, and I would think it would contain all the functionality you would need. Is there something that you saw in the second link that you don't see a way to do using the tools in the first link?

PYmeas, 2020-06-25:
Hello 

new into pymeasure using keithley. I am having a similar issue on a K2420 :AttributeError: 'Keithley2400' object has no attribute 'config_current_source', I am using RS232/USB connection, working fine on some piece of code but for complex acquisition it s troublesome, any idea if an updated version of the code or the keithley class has been updated?
thanks

bilderbuchi, 2020-06-26:
Yes, this is a left-over documentation bug from an API change, `config_current_source` should be replaced with `apply_current` (attention: order of args changed). Can you submit a fix of the docs?

bilderbuchi, 2020-06-26:
Same with the `set_buffer`-> `config_buffer` change you identified in #197

PYmeas, 2020-06-26:
Thanks for the feedback. I got these 2 right but the tutorial still not running properly having some trouble in the current loop error displayed
sourcemeter.current = currents[i]
AttributeError: can't set attribute

also the source seems not enabled 

anyone familiar with this type of data acquisition, i am novice in python.

PYmeas, 2020-06-26:
at the same time while running the code the Keithley 2420 displays error : ID = 800 and under illegal with storage active.

bilderbuchi, 2020-06-26:
Unfortunately, the Keithley24x0 is a rather complex device. In your first case `sm.current` is only for _reading_ the current the device detects (it is a "measurement" in Pymeasure parlance, as opposed to a "control" (https://pymeasure.readthedocs.io/en/latest/api/instruments/instruments.html#pymeasure.instruments.Instrument.measurement).
To set the source quantities, you need to use the `source_voltage` or `source_current` properties, as appropriate (if it's in the correct mode - use apply_voltage/current. Then you enable the output with `enable_source()`.
Further than that you will have to state the exact sequence of commands you use. The example in the docstring of the device should work and give you hints, if this gives errors please report that.

PYmeas, 2020-06-29:
I managed to clear the code to acquire an I-V on a Keithley 2420 and plot the data. there were a few old commands utilized and too many buffer related commands conflicting. The enable source command was initially missing. Below a running code.
``` ruby
# Import necessary packages
from pymeasure.instruments.keithley import Keithley2400
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt
from time import sleep

# Set the input parameters
data_points = 50
averages = 50
max_current = 0.01
min_current = -max_current

# Connect and configure the instrument
sourcemeter = Keithley2400("ASRL6::INSTR")  #using RS 232 COM6
sourcemeter.reset()
sourcemeter.use_front_terminals()
sourcemeter.measure_voltage()
sourcemeter.apply_current()
sourcemeter.source_current = min_current  # Sets the source current to min current mA
sourcemeter.enable_source()
#sourcemeter.config_buffer(averages) - this command is not necessary
sleep(0.1) # wait here to give the instrument time to react

# Allocate arrays to store the measurement results
currents = np.linspace(min_current, max_current, num=data_points)
voltages = np.zeros_like(currents)
voltage_stds = np.zeros_like(currents)
sleep(0.1)

# Loop through each current point, measure and record the voltage
for i in range(data_points):
    sourcemeter.start_buffer()
    sourcemeter.source_current = currents[i]
    sleep(0.1)
    # Record the average voltage and standard deviation
    voltages[i] = sourcemeter.voltage
    voltage_stds[i] = sourcemeter.std_voltage

# Save the data columns in a CSV file
data = pd.DataFrame({
    'Current (A)': currents,
    'Voltage (V)': voltages,
    'Voltage Std (V)': voltage_stds,
})
data.to_csv('example.csv')

# Plot the data with the standard deviation as error bar, note in this example yerr=0.01 to be visible take your voltage_stds in your experiments
plt.errorbar(currents, voltages, yerr=0.01, fmt='.k')
plt.ylabel('Voltage (V)')
plt.xlabel('Current (A)')
plt.title('I-V plot')

sourcemeter.shutdown()
``` 
[image]

Bazabarimailo82, 2021-01-28:
Hello PYmeas, 
Thank you for the update here. I used your code presented here for my Keithely 2450 model. It shows no issues with code but instrument displays two errors 'Error -113' and 'no trigger model found'. Any Idea how to fix this?

PYmeas, 2021-01-29:
@Bazabarimailo82 the trigger model not found is related to this line sourcemeter = Keithley2400("ASRL6::INSTR")  #using RS 232 COM6 cross check you have the right pyvisa model there. I guess it  s different from 2410.

bilderbuchi, 2021-01-29:
Yeah, I would try with the Keithley2450 class, instead. If you encounter problems, try to use current pymeasure master, there have been some recent changes that are not in a released version, yet.

Bazabarimailo82, 2021-01-29:
Hi
Thank you for the information. I used a USB port instead. sourcemeter =
Keithely2450 ( "USB0 :: 0x05E6::044444434::INSTR") was used as provided by
NIMax quarry. Any suggestions?
Thanks
*Narendra Acharya, **PhD*
*Edinburgh, UK*
*Phone: +44-7825616708*

*https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en
<https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en>*

bilderbuchi, 2021-01-29:
> try to use current pymeasure master, there have been some recent changes that are not in a released version, yet.

Bazabarimailo82, 2021-01-29:
Can you suggest the link for  'current pymeasure master'  please?
*Narendra Acharya, **PhD*
*Edinburgh, UK*
*Phone: +44-7825616708*

*https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en
<https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en>*

bilderbuchi, 2021-01-29:
Please see the documentation at https://pymeasure.readthedocs.io/en/latest/dev/contribute.html#using-the-development-version (you can disregard the part about forking to your account, and use https://github.com/pymeasure/pymeasure.git instead.

Bazabarimailo82, 2021-02-01:
Hi thank you for the information. I used the link provided (
https://github.com/pymeasure/pymeasure/tree/master/examples/Current-Voltage%20Measurements)
and ran the example code. Python execution shows an error as follows:

voltage = self.meter.voltage

AttributeError: 'IVProcedure' object has no attribute 'meter'

The code in the link thas two source meters used however I used only one
sourcemeter (2450) with USB connection for two points measurement and
changed interface address accordingly.

Can you help me to solve this please?

Thanks
*Narendra Acharya, **PhD*
*Edinburgh, UK*
*Phone: +44-7825616708*

*https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en
<https://scholar.google.com/citations?user=fG-XekEAAAAJ&hl=en>*

bmoneke, 2023-06-05:
@PierreLottigier which pymeasure versión do you have?
Could you try some more basic actions like getting or setting a voltage?

And what does the manual state regarding that error code you got on the screen?

The error message (in the python program) indicates, that the device did not respond in time. Now we have to find the reason, for which the message on the screen is the best hint.

PierreLottigier, 2023-06-06:
So I think the problem is higher, since I cannot communicate with my instrument even in Keithley's communication I/O soft ...

PierreLottigier, 2023-06-06:
The problem was that pymeasure is sending SCPI commands to the instrument.
So to make it compatible it is needed to set up the Keihley accordingly (which by default is in TSP mode): 
Menu > System Settings > Command Set > SCPI